motioneye-project/motioneyeos #2365: Starting wpa_supplicant: failed (r8188eu).

sarathchandars, 2020-04-26:
Got this problem today with 20190911 release on my Pi Zero (not W) + TL-WN725N

Error message: Starting wpa_supplicant: failed (r8188eu). System would go into a loop restarting and trying.

starbasessd, 2020-04-26:
Does your WiFi dongle work on the PiZero with a basic Raspbian?
Also try the motionEyeOS 20200203:
https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi-20200203.img.xz

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AstroSmoke, 2020-04-27:
Hi,
I'm having a similar issue with a Raspberry Pi3A+ where It will endless loop fail to engage the wireless, after adding a wpa_supplicant conf
Using motioneyeos-raspberrypi3-dev20200223.img

starbasessd, 2020-04-27:
Have you tried it with your Pi3a and Raspbian 20200213?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

sarathchandars, 2020-04-27:
motionEyeOS 20200203 worked out of the box with the same wpa_supplicant.conf file from yesterday. Thanks!

closed 2020-04-27 by sarathchandars