motioneye-project/motioneyeos #2366: Server loses connection with clients at the same time? Errno 113 / Warn 403

RobertKrens, 2020-04-28:
First of all, I have this setup working for more than a year and love it!

Setup:
 - RPI Zero + V2.1 camera – Fixed IP xxx.xxx.xxx.101 – Wifi connected
 - RPI Zero + V2.1 camera – Fixed IP xxx.xxx.xxx.102  – Wifi connected
 - RPI 3B+ = MotionEye Server – Fixed IP xxx.xxx.xxx.99 – Wired connection to router
 - RPI 3B+ TouchScreen = Raspbian Stretch -> Fixed IP xxx.xxx.xxx.60 has Chromium opening IP xxx.xxx.xxx.99 (server) – Wireless connection

MotionEye Version 0.41
Motion vesion 4.2.2
OS Version 20190911

The setup works well for most of the time, but almost every day it loses connection to both camera’s at the same time. For this I think there is something wrong with the Server, but cannot seem to find why it does this. The MotionEye log gives these two errors (multiple times):

2020-04-27 00:40:57: [motioneye]  WARNING: error on read: [Errno 113] No route to host
2020-04-27 09:07:41: [motioneye]  WARNING: 403 GET /login/?_=1587971262391&_username=_&_signature=32adf11eed301ae7dff2d01728d61f5d48dd410c (192.168.1.60) 4.79ms

The messages has the following:
Apr 27 11:34:34 Claespolderstraat user.warn motion: [1:ml1:Voordeur] [WRN] [NET] Above message repeats 2 times
Apr 27 11:34:34 Claespolderstraat user.warn motion: [1:ml1:Voordeur] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
Apr 27 11:34:34 Claespolderstraat user.warn motion: [2:ml2:Schuurdeur] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
Apr 27 11:34:34 Claespolderstraat user.warn motion: [1:ml1:Voordeur] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd

The messages shows an undervoltage sometimes
Apr 27 11:33:37 Claespolderstraat user.notice throttlewatch: currently: under-voltage throttled 
Apr 27 11:34:07 Claespolderstraat user.notice throttlewatch: currently: under-voltage throttled

So I’m not sure where to look for a solution. 
Any help is appreciated. Thanks in Advance!

Regards Robert

[image]

starbasessd, 2020-04-28:
Does it happen at the same time every day?
Can you try changing the power supply on the .99 machine (if that is where
you are seeing the under-voltage error)?
Where is the .99 machine? In a cooled environment? Sunny window? Good
airflow? Bad airflow?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

RobertKrens, 2020-05-01:
> Does it happen at the same time every day? Can you try changing the power supply on the .99 machine (if that is where you are seeing the under-voltage error)? Where is the .99 machine? In a cooled environment? Sunny window? Good airflow? Bad airflow?
> […](#)

- No it happened at random. 

 - I changed the power wire and it seems a faulty wire. Hasn't done it ever since. I feel a bit stupid for underestimating the power failure. 

 - The .99 machine is in next to the fuse box. It's room temperature plus some heat of the Pi itself. The Pi-Hole next to it says around 40degC

Thanks for the reply, think the topic can be closed.

Regards Robert

closed 2020-05-01 by RobertKrens