motioneye-project/motioneyeos #2371: Mac terminal ssh pi

Ahaso, 2020-05-05:
Im not very familiar with these stuff. Have to say that motioneyeos is very good tool. I have couple things what I'm need to do using ssh protocol.
Trying to connect pi using macOs terminal. Im typing ssh pi@xxx.xxx.xxx.xxx  
Welcome to meye-1f32e74a! 
pi@xxx.xxx.xxx.xxx's password: 
What is the default password?

starbasessd, 2020-05-05:
it's
ssh admin@xxx.xxx.xxx.xxx
or
ssh root@xxx.xxx.xxx.xxx
and if you haven't changed it in the web page it's blank (but should let
wyo uin without entering anything.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Ahaso, 2020-05-05:
Thanks, I'm feel stupid now.

closed 2020-05-05 by Ahaso