motioneye-project/motioneyeos #2372: Cannot Access Interface

PolarBear456, 2020-05-07:
I can see the IP of my Pi Zero with MotionEyeOS as 192.168.1.13 on my route page. When I enter this in any browser, I get no result. Never stops loading. If I do ping 192.168.1.13 in Command Prompt, I get "destination host unreachable".

starbasessd, 2020-05-07:
Can you ping the IP address?
What IP address does the device you are trying to connect to it with have?
Phone, Tablet, Windows PC, MAC, Linux?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PolarBear456, 2020-05-07:
I have tried on an iPhone, Windows PC and another Windows PC. The IP is 192.168.1.13. This is displayed in the router preferences on my router settings as meye-... . When I try to ping over command prompt, I get "destination host unreachable". When I try to enter this in a browser, the page will not load ("TIMEOUT").

starbasessd, 2020-05-07:
Can you connect a monitor to the PI? What does it show?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-07:
Also, which image are you running?
Can you try the one from this link:
https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi-20200203.img.xz

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-05-07 by PolarBear456

PolarBear456, 2020-05-07:
Yes, thanks for your help. If I connect it to a display, it showed that the IP was 192.168.1.8. Turned out I was entering the wrong IP!

starbasessd, 2020-05-07:
Good deal, and good luck!

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?