motioneye-project/motioneyeos #2374: Boot EyeOS from USB HDD

jangix, 2020-05-10:
I am using the following board/model: Raspberry PI 3B+.
Is it possible to boot the Motoneye OS from USB external magnetic hard disk? (without sd card)
How can I do that?

starbasessd, 2020-05-10:
You can't unless you build your own. Instructions here:
https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source
I can't even get the bootcode.bin workaround to work...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jangix, 2020-05-10:
I don't know what should I modificate to make the boot from HDD work, I even don't know if it's possible.
It would be perfect if someone more advanced could publish a compiled version that allows that feature.
I think it would be very useful for the other users that use motioneye as NVR with the countineous videorecording.

> You can't unless you build your own. Instructions here: https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source I can't even get the bootcode.bin workaround to work...

starbasessd, 2020-05-10:
I've been trying for a while. It may be a limitation of the BuildRoot
system CCrisan uses.

My work-around is to save to a SMB (Network Share) server. or to the Custom
Path mounted /data/media/sda1 USB drive, but still running everything from
the mEOS SD Card.
If you have the USB Drive plugged in, formatted to exFAT or Fat32, and the
SDCard running motionEyeOS
Go to File Storage
change the Path to /data/media/sda1
Apply at the top
Files will now be stored on the USB drive
Remember to shut down the PI before removing or inserting the USB Drive.

Another work around is to do a Rasbian install to USB, and then install the
motionEye stuff from these instructions:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jangix, 2020-06-01:
Oh good we are on the same page!
I tried the same 3 solutions which you proposed but I didn't resolve it (recording folder into mounted external HDD or into SMB shared folder or using MotionEye on Raspian).
The common limit of this ways is the unmanaged fault error during the saving of the video registrations.
When writing data to the destination path fails for any reason*, the operating system doesn't stop recording but continues to save it on the SD card, instead on the HDD or SMB Server (at the same relative path).
This problem causes the saturation in a short time of the 8GB of the SD card and the total blocking of the recordings and a down of the system performance.
Cleaning the system often requires long and very annoying maintenance via SSH.
The same problem when it happens on MotionEye in Raspian, it completely blocks the boot of the operating system.

*USB disconnection of the HDD, overheating, memory filling, HDD writing error for bad sector, restarting the SMB server, etc.

PsyEng, 2020-06-21:
If you like to boot from external devices, look into my small guide 
[Boot motioneyeOS from USB/SSD on Raspberry Pi 4B only](https://github.com/ccrisan/motioneyeos/issues/2439)