motioneye-project/motioneyeos #2375: MotionEyesOS Short Video Clips Every Hout

bellzbubba, 2020-05-10:
### Preliminary Docs

Note: This technically is NOT an Issue.  It is a request for help with a unique question.

I just configured MotionEyesOS on a Pi Zero W for a singular and somewhat different purpose.  I wish to make short (5-10 second) video clips at regular intervals.  I'm thinking hourly, but that may change.

I have everything else working.  It takes pictures every hour and sends them to my server.

And I have poked around here and elsewhere for tips.  I "think" that I can accomplish what I want using a combination of CRON and a Python script ... but I am NOT familiar with either, or the OS.

ANY help would be immensely appreciated.

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
I am running motionEyeOS version: (insert your version here, e.g. 20190911).
 motionEye Version: 0.41
 Motion Version: 4.2.2

### Board Model
I am using the following board/model: Pi Zero W.

### Camera
I am using the following type of camera: MMAL.
My camera model is: Raspberry Pi Camera V2.

### Network Connection
My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals
I am using the following peripherals that I consider relevant to this issue: NONE

starbasessd, 2020-05-10:
You may want to consider running Raspbian Lite, and follow these
instructions:
https://www.raspberrypi.org/documentation/usage/camera/raspicam/timelapse.md
Doesn't require python, and seems to work for both jpegs (pictures) and avi
(movies)...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bellzbubba, 2020-05-10:
I don't really want "Time Lapse" video.  I want 10 second "Clips" every hour.  Totally different ...
But thanks for the input.

starbasessd, 2020-05-10:
Look at the bottom section.
It records a short AVI, and if you use crontab to do the recording, you get
what you were asking for.
You can preset where the file is saved, even to a network share.
And no, it's not 'totally different'. Just a matter of semantics...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bellzbubba, 2020-05-10:
Well I don't want to UNDO everything I've done just to install Raspbian Lite.  And the "raspvid" and "raspistill" commands generates the same error of "mmal: mmal_vc_component_enable: failed to enable component: ENOSPC" and have no clue what that means.

starbasessd, 2020-05-10:
The error means you forgot to enable the camera in Raspbian. No big deal.
sudo raspi-config
Interfaces
enable camera
motionEyeOS auto-detects/enables the camera.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bellzbubba, 2020-05-10:
Yes, my camera works fine.  But I AM running motionEyesOS and am NOT switching to Raspbian Lite.