motioneye-project/motioneyeos #2377: 5 camera cctv log system

BarRae, 2020-05-12:
Hello,

I have a general question.
We want to develop a 5 camera machine surveillance system.
This system has to record video from 5 camera and save it to 1 central location.
It has to do this for 3 days straight ideally at 1080p in 120 fps.
Would this be possible using motioneyeos and a development board (like raspberry pi)?

If you have any other remarks or suggestions feel free to tell me.

starbasessd, 2020-05-12:
I doubt it. RPi & MotionEye struggle with 5-10 FPS with 5 cameras due to
several limitations, including bus bandwidth, CPU (ARM SoC), etc.
Other boards may be more amenable, but the ARMs probably won't do what you
are looking for.
You'd be hard pressed to do that level of video over Ethernet without some
form of compression at the camera (H.264 or similar)
It's be almost impossible to do it over WiFi, reliably.
You might want to consider some form of motion detection to cut back on
bandwidth (say, 5-10 FPS with no motion detected, and burst it up to the
120FPS when active.
Storage is not an issue, especially on a Pi4 with a USB external drive
(SSD/HD/Thumb Drive).
If you can find an old SWANN camera system, open it up, and see how they do
it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BarRae, 2020-05-12:
What would be the maximum resolution at like 60 fps using RPi and motioneyeos then?

starbasessd, 2020-05-12:
I can't get even 800x600 at 20 FPS for 5 cameras.
That is still a lot of bandwidth at several levels. RPi pre 4 were limited
to about 300Mbps due to the USB2 bus shared with ethernet, and 56Mbps on
G/N wifi.
I don't think an RPi family is going to be your solution. You could try a
NUK, or Intel processor based product, but the cost is going to be a lot
more. They have linuxes for it, and most linuxes support motionEye or can
compile it for them.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BarRae, 2020-05-12:
We do have an aaeon up squared boars which is intel processor based board I take away that this should definitely be the more capable board for this then?

starbasessd, 2020-05-12:
No clue. I am an RPi guy, and AMD. Although BillG and Intel keep me
employed, I prefer Linux and AMD and RPi.
It sounds like you are trying to do a commercial system (1080p/120FPS
cameras aren't cheap to begin with), true?
Are you doing it to replace one you have currently, or trying to develop
one for the marketplace?
Just curious. You'd need a programmer to optimize for whatever hardware you
end up with.
I'd advice a good gpu, too.
I might suggest a refurbed PC, a good Linux distro, and decent cameras.
I use a Lenovo MT M 10AY tiny ThinkCenter for my day-to-day meOS PC (Ubuntu
20.04/8GB/1TB/GBit Ethernet/-nWiFi/Core i5)
and I don't get anywhere near the specs you are looking for. I think you'd
need a system designed from the ground up for that.
I've only seen systems like that in a casino I worked at, and the NVR used
was still not getting what the security team wanted.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-12:
From my Lenovo Camera Server - 20/25 FPS total. shared among the 5 cameras.
4 IP Cameras (3 D-Link 5020L, 1 PiZero USB Cam) & 1 Direct Connect USB Cam
[image: image.png]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BarRae, 2020-05-12:
Well, yes and no. I work for a firm developing custom solutions for customers. The system in question will go to a customer but will be a one-off thing.

starbasessd, 2020-05-12:
Here's one example (this one does 8 channels (coax) has the spec of
1080p/120fps, but it is referring to total fps, so if you used all 8
channels that would be 15 fps each.)
https://www.worldeyecam.com/full-hd-120fps-1080p-hd-sdi-4-channel-dvr.html
and it's $800 without cameras or HDs.
To get 120FPS per channel (even at only 5 channels, you'd need bandwidth at
every level to support 600fps, and you're talking tens or even hundreds of
thousands of dollars.
It might be cheaper to go with 5 separate sub-systems. Or reset your
client's expectations of 120 FPS per channel into a single system.
I could see the PI acting as a hub for 5 systems capable of 120fps per,
have the 5 systems maintain their own video and just use the PI as a
supervisor.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BarRae, 2020-05-13:
Hello, I have one more question,
Would you happen to know what the maximum expected framerate would be at 1080p when using a single camera per Raspberry pi.
So 5 camera's and 5 RPi's

closed 2020-06-05 by BarRae