motioneye-project/motioneyeos #2378: New stable release?

roger-, 2020-05-13:
Any idea when there'll be a new stable release?

starbasessd, 2020-05-13:
I've been using 20200203 for a while (since Mid February) and it is very
stable, if not listed as such, and the one I recommend to use if you see
any issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JohanD78, 2020-05-13:
Anything after 20190911 breaks continuous recording. (First segemt records but nothing after that)

Waiting for new stable and /or devbuild with fixes from motioneye that corrects that issue.

We will just have to wait and hope Calin updates it soon :)

sts-ryan-holton, 2020-05-13:
@JohanD78 

> Anything after 20190911 breaks continuous recording.

I'm getting the exact same issue, with continuous recording turned on I get the first "video" recorded, and then it just stops recording after this, I've had to resort to using photos taken every 10s for instance, but this isn't great!

Any workaround for this or fix coming soon?

@ccrisan