motioneye-project/motioneyeos #2381: cant upgrade to 20190911 space full

cr0co, 2020-05-14:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190119

### Board Model

I am using the following board/model: rpi zero w

### Camera

I am using the following type of camera: V4L2

My camera model is: 

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

### Log Files

I consider the following log files relevant to this issue:

fwupdate upgrade 20190911
upgrading to 20190911
downloading...
downloaded 20190911
extracting...
not enough space on root partition (230686720 needed, 209715200 available)
[root@Mobil ~]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               193.7M    168.9M     10.8M  94% /
devtmpfs                180.8M         0    180.8M   0% /dev
tmpfs                   185.0M     16.0K    185.0M   0% /dev/shm
tmpfs                   185.0M      2.3M    182.7M   1% /tmp
/dev/mmcblk0p1           29.9M     10.5M     19.4M  35% /boot
/dev/mmcblk0p3           13.6G    429.4M     12.4G   3% /data
/dev/mmcblk0p3           13.6G    429.4M     12.4G   3% /home/ftp/sdcard
/dev/mmcblk0p3           13.6G    429.4M     12.4G   3% /home/ftp/storage

how to fix this issue?

starbasessd, 2020-05-14:
What version are you upgrading from?
Several upgrade paths are not available, due to partition structure changes.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

cr0co, 2020-05-14:
I am running motionEyeOS version: 20190119

starbasessd, 2020-05-14:
Hmmm. I would save the configuration, save any files (pics and movies) to
another drive, and re-image..

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?