motioneye-project/motioneyeos #2382: Motion setup bar in web gui missing

macdroid53, 2020-05-14:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

Image: motioneyeos-raspberrypi4-20190911

### Board Model
 Raspberry PI 4).

### Camera

Network Camera, Simple MJPEG Camera).

FOscam FI8918W

### Network Connection

Ethernet

When I access the web gui I get no motion setup bar. I've not found any mentione of missing motion setup in the gui...so don't know what I've missed.

Probably my non-creative ideas for search terms...

starbasessd, 2020-05-14:
Are you logged in as Admin?
Click the profile in the upper left, and sign in as admin.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-14:
I think I'm logged in as admin...I know of no other user to log in as...but, I logged in again...no difference.

starbasessd, 2020-05-14:
So when you log in you don't see:
[image: image.png]
The third from the bottom bar?
Can you screen shot what you do see?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-14:
I cant see the image you posted, this is what I see (I closed them all down):
[image]

starbasessd, 2020-05-14:
Turn off "Fast Network Camera" under Expert Settings.
Fast Network Camera only works with a camera (like the "Official" one from
RPI that connects to the CSI connector) and disables a lot of functions.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-14:
It was off.
I turned it on, the Pi rebooted.
bunch of other stuff showed up under Expert 
turned fast cam off, reboot
still no motion setup

starbasessd, 2020-05-14:
Confirm that you are running the image:

https://github.com/ccrisan/motioneyeos/releases/download/20190911/motioneyeos-raspberrypi4-20190911.img.xz

If needed, please re-image with this version or

https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi4-20200203.img.xz

I have confirmed that both are working on my Pi4-4GB

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-14:
From the terminal on the pi I can still use reboot. But, it says shutdown is not there, what's the shutdown for the command line? (for example I tried: shutdown -h now)
It is currently running ...pi4-20190911

starbasessd, 2020-05-14:
Try
halt <enter>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-14:
I re-imaged with the following:
sudo ./writeimage.sh -d /dev/sdb -i "motioneyeos-raspberrypi4-20200203.img"

Still no motion setup bar.

starbasessd, 2020-05-14:
Linux or MAC? I'm getting to create a new SD Card for my Pi4-4GB.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-14:
linux

starbasessd, 2020-05-14:
Back in a few minutes.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-14:
Rpi4-4GB, 16GB SD Card, wpa_supplicant for my network & ssh copied to /boot
partition
Ethernet & WiFi connected

https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi4-20200203.img.xz
Network Camera dLink DCS-5020L
Image SD Card.
copied files to /boot partition
unmount and insert into PI4
turn on
watch for IP address
connect to WebGUI on IP Address
Log in as admin
set up network camera
Get video from camera, no other settings changed
[image: image.png]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-14:
In case image didn't display, all 14 lines / blocks are there

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-14:
Can't see the image.

what does writeimage.sh do?

The first time I did a dd to my 32G sd card it wouldn't boot.
That's why I got writeimage.sh from git.

When it completes the card has 2 partitions. I have not copied any files, by hand, to boot.

starbasessd, 2020-05-14:
When I look at the partitions, after a setup boot, I get 5 partitions: (3
active)
boot (30MB)
unallocated (69MB)
root (220MB)
unallocated (704MB)
data (13.46MB)
on a 16GB card

writeimage.sh, when push comes to shove, does a 'dd' (

dd if=$DISK_IMG of=$SDCARD_DEV bs=1048576

)
with the possibility of a bunch of options.
on my linux system I just do the dd.directly
On my Windows machines, I use Win32DiskImager
I haven't done it on my Mac.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-14:
Well, I've now used dd, writeimage.sh, and mkusb and got the same results each time:

A system that can see the video from a camera, but offers no motion setup and 7 blank bars in the left side (i.e. no file storage, text overlay, etc.)

starbasessd, 2020-05-14:
Did you try re-downloading from github?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-14:
Downloaded multiple times in including the link in this thread. After decompressing they are around 60mb. Seems small, but I have no reference.

starbasessd, 2020-05-14:
Should be in the neighborhood of 337MB (337,920 KB) for the uncompressed
img file.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-14:
Ok, I got it from here: https://github.com/ccrisan/motioneyeos/releases

And the .xz is 56.1mb and the extracted .img is 346,030,080 bytes.

I did a dif with the writeimage.sh and what I had gotten from somewhere in the wiki and there were some dif's.

So I tried writing the image again. No change, no motion setup.

macdroid53, 2020-05-14:
Just to confirm, if it can see the feed from the camera, it should be able to do motion?

I.e. the type of camera does/doesn't change motion setup being available...

starbasessd, 2020-05-14:
Do you have a different SDCard to try?
If not, I really don't know what your issue is.
Are you able to watch the boot with a monitor?
You can try attaching all 5 log files to here, too, and maybe @ccrisan
might have an idea

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

macdroid53, 2020-05-14:
There is no motion.log.

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4631276/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4631277/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4631280/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4631281/motioneye.log)

macdroid53, 2020-05-14:
Just tried a different sd card, no change

macdroid53, 2020-05-15:
So, summery:
- tried 2 different versions of the image
- tried 3 different ways of writing the image
- tried initial format of SD card ext4 and fat32
- tried 2 different SD card writing devices

I get a camera feed (simple mjpeg network cam) and 7 of the function bars blank (i.e. no motion setup, no email setup, etc.)

This is a known good R Pi (and only a few hours on it...).

What am I doing wrong?

ccrisan, 2020-05-16:
@macdroid53 if you add cameras as Simple MJPEG Network cameras, you won't get any motion detection parameters. Please use "Network Camera" when adding your camera.

closed 2020-05-16 by ccrisan

macdroid53, 2020-05-17:
I'm sure you can point me to what I missed about this in the wiki or elsewhere, but, I DID do a lot of searching and found nothing that discussed this. Could you provide a link?

ccrisan, 2020-05-19:
It's not in the Wiki, but written directly in the UI. See the description below:

[image]

macdroid53, 2020-05-19:
Well, shame on me. I looked at that and never connected "but no motion detection" with "several panels will be blank".

Sorry, I'm really bad at extrapolating things like that into what it is implying. Of course once pointed out what the connection is, it's obvious now.

starbasessd, 2020-05-19:
Thanks, @ccrisan for that info. I hadn't paid attention, and I've been
trying to support your product for a while. I love learning things, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?