motioneye-project/motioneyeos #2386: Mutiple devices - Grey Screen

tannaroo, 2020-05-17:
Hi - newbie here and have read the FAQ etc and forum but still can't seem to get my 2 cameras to work on MotioneyeOS as I have the dreaded gey screen.

I am running motionEyeOS version: 20190911 on a RPi 4 which is connected to my network via ethernet.

I am using the following type of camera: Reolink RLC 410 and HiVision ColorVu as a network camera using their respective rstp/ip settings with the username/password boxes filled in the 'motioneye add a camera setup' but get the grey screen / unable to access video.

For example, the Reolink is setup as: rtsp://<ip address:port>. Tried to different configurations from reolink support but all get the same grey video screen.

I can access both cameras using their respective IP addresses through a web browser.

Any help greatly appreciated

tannaroo, 2020-05-18:
Do I need to install ffmpeg to get this working?
Or should I just install Raspian then motioneye rather than motioneyeOS?

starbasessd, 2020-05-18:
How are you installing them?
What settings are you using?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
To date, I have just installed MotioneyeOS following the guide/wiki which doesn't say anything about ffmpeg (I only got that via that raspian/motioneye forum).

So I have followed the MotioneyeOS and I can access the web interface and add both cameras (Reolink and Hikvision) but the video is grey/unable to connect issue, so no tsure what the issue is?

starbasessd, 2020-05-18:
What settings do you put in when you add a camera?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
Add a network camera:
rtsp://<IP addess>:554 (also tried: rtsp://<IP addess>:554/Streaming/channels/101
username: <my username>
password: <my passowrd>
tried both tcp/ucp

Similar to the Reolink camera.

starbasessd, 2020-05-18:
> Add a network camera:
> rtsp://:554 (also tried: rtsp://:554/Streaming/channels/101
> username:
> password:
> tried both tcp/ucp
>
> Similar to the Reolink camera.

For the Reolink try:
Delete the camera
add camera, Network
URL:  *rtsp://admin:admin@<ipaddress>/h264Preview_01_sub*

*Let me know.*

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-18:
Sorry, oops on my part:

URL:  *rtsp://<ipaddress>/h264Preview_01_sub*

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
Thank you thas has worked!
Would have the the correct URL for the Hikvision as the one I had was from Hikvision forums but maybe its wrong?

starbasessd, 2020-05-18:
Try:

rtsp://<ipaddress>:10554/Streaming/channels/101

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-18:
BTW, Google is your friend:
Search Terms (manufacturer) (model) "rtsp url"
Lots of sites even have 'plug in your info' url generators.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
I know I have been googling for days and tried many different url including the one you have provided.
rtsp://<ipaddress>:10554/Streaming/channels/101
but it doesn't work. but thanks anyway. i at least have one working.

starbasessd, 2020-05-18:
> rtsp://:10554/Streaming/channels/101

With or without authentication?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
tried both

starbasessd, 2020-05-18:
Which is the camera set for?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
Its with authentication now

starbasessd, 2020-05-18:
Which model specifically?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
Its the HiVision ColorVu: DS-2CD2347G1-LU

starbasessd, 2020-05-18:
Try
Try: rtsp://<ipaddress>:10554/Streaming/channels/102   <<< note the 102
with your username & password.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-18:
Also try using VLC to see if that program can access.
Looks like they couldn't pick a system and stick to it, each model is
different...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
Ok the 102 channel also did not work. By the way, it wont accept port 10554 (says connection refused), so used 554 which gets me back to the grey screen.

Will need to educate myself and try it in VLC

starbasessd, 2020-05-18:
Try: rtsp://<ipaddress>:554/Streaming/channels/101
(That's from the manual for that camera)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
Yes that is the one I started with and it didn't work.

starbasessd, 2020-05-18:
Manual for the camera is at
https://www.hikvision.com/en/products/IP-Products/Network-Cameras/Pro-Series-EasyIP-/DS-2CD2347G1-L-U-/
Make sure you start with a factory reset before making camera changes, so
it's clean
Once you make it clean and are able to connect, then make your other
changes one by one.
Good Luck.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-18:
Did you try 102, also? (Secondary stream)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
Yes.
Its probably a setting on the camera, so I will try the factory reset option and try again

starbasessd, 2020-05-18:
Also check for latest firmware.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
I just tried the rtsp stream in VLC as I used it in motioneye and it worked in VLC but when I try it in motioneye - it doesn't work.

starbasessd, 2020-05-18:
Interesting. The same settings should work in both...
Do the setup from:
https://www.hikvision.com/content/dam/hikvision/en/support/download/how-to/ipc/How%20to%20Get%20IPC%20MJPEG%20Stream%20via%20HTTP%20Commands.pdf
then
Try http://<ipaddress>/ISAPI/Streaming/channels/102/httpPreview

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-18:
Still using the Network Camera setup

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
Did those changes and when I try to add: http://ipaddress/ISAPI/Streaming/channels/102/httpPreview

it says authentication failed. I tried to also add the port number per my camera web portal for http and that also says authenticiation failed.

starbasessd, 2020-05-18:
Did you put in the admin & password, or did you set the camera to no
user/password?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
I tried to insert the admin & password within the url and also without it but on the separate input boxes on network camera setup bu it says authentication failed in both cases.

meanwhile on my other laptop, i have that camera stream on vlc working throughout.

starbasessd, 2020-05-18:
I really don't know, and don't have one.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tannaroo, 2020-05-18:
That is fine. Really appreciate your help. You got my reolink working and taught me how to use VLC for camera streams as well so that is a plus.

Quick question and I know I should google this and will probably find the answer but I want to buy a Eufy Outdoor Camera and Eufy Video Doorbell - do you know if these streams can be viewed into motioneyeOS?

starbasessd, 2020-05-18:
Maybe:
http://community.anker.com/t/how-to-setup-rtsp-on-eufycam-to-stream-video-to-your-compatible-nas/64833

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

douglaslwarren, 2020-05-22:
tannaroo,

I'm having the same problem with a HIKVision DS-2CD2322WD-I turret camera.   I can see it in VLC media player on both the Pi and my windows machine....but can't get MotionEye to see it when I add the IP address...