motioneye-project/motioneyeos #2389: Argon one Pi 4 fan script with motioneyeOS

fr33soul, 2020-05-18:
### motionEyeOS Version 20200203
### Board Model Raspberry Pi 4

I bought an Argon one Pi 4 minicase with a fan
and in orther to enable this fan i have to lunch this script on the terminal:

`curl https://download.argon40.com/argon1.sh | bash`

but it returns `sudo not found`
if i remove sudos from the sh it returns nothing..
i need this script..

[argonpower-script.txt](https://github.com/ccrisan/motioneyeos/files/4647319/argonpower-script.txt)

how can i achive that?

starbasessd, 2020-05-18:
curl doesn't exist in motionEyeOS and there isn't an easy way to install it.
There are a number of other commands that don't exist, either that are in
the argon1.sh script

You could run Raspbian and then install motionEye.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-05-18:
BTW, my test procedure:
Clean image of motionEyeOS to 8GB SDCard
Pi4-4GB
copied ssh.txt and wpa_supplicant.conf to /boot partition
insert SD Card to Pi4 & power up
Wait for prompt to connect via browser
connect to ipaddress via ssh
cd /data/media (use as temp folder for testing
run curl https://download.argon40.com/argon1.sh
'command not found'
run nano ./argon1.sh, copy from another computer and paste into here
run bash ./argon1.sh
various errors, missing applications, get warning to try to download again
rm ./argon1.sh
scp user@anothercomputer:/share/argon1.sh ./
run bash ./argon1.sh
same errors

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

fr33soul, 2020-08-08:
well.. it is a serious problem, as cooling systems are very important especially for video procedures that heat up the hardware a lot .. is it possible to consider this very important problem?

starbasessd, 2020-08-08:
There are a lot of  devices not supported. mostly because they are not
supported directly by the Raspberry PI OS, either. I might suggest using
the alternate install by installing RaspberryPIOS, then install
motion/motioneye on it, then installing your Argon1 software there, where
it might be better supported.
I know the PiMoroni fan shim also isn't supported, easily, either, and it's
my work-around. You could also set the fan to be direct powered, always on
by wiring it direct to the 5v power pins on the PI.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?