motioneye-project/motioneyeos #2392: cameras are frozen

holzfelix, 2020-05-23:
### motionEyeOS Version

motion eye version 0.41

### Board Model

Raspberry PI 3B+

### Camera

POE IP cam Belko Full HD - 4 Cameras same type

### Network Connection

Ethernet

### Peripherals

nothing. data store is on a nas

I often have the problem that the cameras hang up. Motion is running and drawing on the camera but the picture is frozen. what could be the reason?

starbasessd, 2020-05-23:
Which version of motionEyeOS? begins 2019 or 2020 or it is very old.
How is the Pi3B+ powered and connected to your network?
NAS using FTP or SMB services?
When you say 'hang up' what do you mean, specifically? Does the stream
simply hang/stop updating? Go to an icon of a camera with the slashed
circle? Go to a grey screen with a last contacted date & time?
When it happens, do you simply power off/on the PI, or do you have to do
more?
PLEASE try to answer all these questions to the fullest of your ability.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

holzfelix, 2020-05-23:
[image]

* Raspberry is powerd with an original power supply. and attached via LAN to my switch (unify)

Network storage:
[image]

* Hang up is a freeze. Camera shows a last image with timestamp when it stopps. 
* I go to motionOS and click restart

Thanks in advance

starbasessd, 2020-05-23:
Can you attach your messages.log, motion.log, and motioneye.log, please?
Can you ssh into the Pi3B+ and run the command
/opt/vc/bin/vcgencmd measure_temp
and report then result? (Checks the CPU Temperature)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

weissjoerg, 2020-05-26:
Hello 
I would like to report the same problem as holzfelix did. I run 2 cameras on my system. 
1. original CSI camera - motion detetection is activated 
2. logitech USB Webcam - no motion detection just video streaming

I'm using a raspberry PI 3B+ which is plugged via original USB power supply. It is connected via WLAN.
This configuration works without any issues for months. 

I recognized app. since 4 weeks that from time to time the USB cam freezes or it is unknown. First of all I thougt it might be an issue of less USB power, so I removed my USB fan but still the same problem. Finally I completely cleaned my system and did a new set up - unfortunately the same issue. 

Either I get a picture freeze on cam 2 or I get no picture - everything works fine after reboot - sometime for a few minutes - sometimes for hours
[image]

I just checked the temperature (without fan) 44,0°C

I hope anyone can help

weissjoerg, 2020-05-28:
screenshot of a frozen cam

[image]

:(
@holzfelix: did you already solve this issue?

rpggodboss, 2020-05-29:
Having the same issue, video streams fine then freezes at some point. No rhyme or reason, but usually occurs in less than 2 hours. I can simply refresh the UI I'm viewing on, and video usually resumes. I don't think it's so much an issue with the camera/pi zero.

starbasessd, 2020-05-29:
No one is supplying requested information to help you.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

rpggodboss, 2020-05-29:
Which version of motionEyeOS? begins 2019 or 2020 or it is very old. : **0.41**
How is the **Pi Zero W** powered and connected to your network? **2.5A usb / wifi 2.4ghz**
NAS using FTP or SMB services? **No NAS here**
When you say 'hang up' what do you mean, specifically? Does the stream
simply hang/stop updating? Go to an icon of a camera with the slashed
circle? Go to a grey screen with a last contacted date & time? **Displays still image** 
When it happens, do you simply power off/on the PI, or do you have to do
more? **I refresh viewer on whatever device I'm viewing on, it does prompt for login info**
PLEASE try to answer all these questions to the fullest of your ability.

A little more about my setup: single Pi Zero W with csi/ribbon cable camera, running stock clocks, set in Fast Network Camera mode. I am not using it to record, just stream over wifi network. I am using it to monitor inside a small shed.

weissjoerg, 2020-05-29:
Which version of motionEyeOS? begins 2019 or 2020 or it is very old. : **0.41**
How is the Pi Zero W powered and connected to your network? **2.5A usb / wifi 2.4ghz**
NAS using FTP or SMB services? **NAS using FTP**
When you say 'hang up' what do you mean, specifically? Does the stream
simply hang/stop updating? Go to an icon of a camera with the slashed
circle? Go to a grey screen with a last contacted date & time? **Freeze as shown in the pictures above**
When it happens, do you simply power off/on the PI, or do you have to do
more? **reboot and afterwards it works for a while**

weissjoerg, 2020-05-30:
I just tried an 3.0A Power Supply  - same problem here

starbasessd, 2020-05-30:
You didn't answer the questions...
Which version of motionEyeOS? begins 2019 or 2020 or it is very old. :
*0.41* << should begin 2019 or 2020, like 20190911 or20200203. If earlier,
upgrade and report
If you have access to console or ssh session, when it hangs, run the
command:

/opt/vc/bin/vcgencmd get_throttled

and report the results. I will bet it's not 0x0 (no  issues)  If it reports
anything else, it indicates a temperature (of the PI) issue or voltage
issue (the number will be specific.
It can be that the PI itself is failing, too.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

weissjoerg, 2020-05-30:
I am using motionEyeOS 20190911 because it is the latest release (no dev).

ssh is no problem, result of /opt/vc/bin/vcgencmd get_throttled
is:  **throttled=0x0**

should i try the dev version? 20200223??

starbasessd, 2020-05-30:
I recommend 20200203, not quite 'bleeding edge'....
https://github.com/ccrisan/motioneyeos/releases/tag/20200203

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

weissjoerg, 2020-06-01:
20200203 installed... same issue here 😑

any further ideas for me?

elmolinger, 2020-06-04:
Same Problem here

MotioneyeOS 20190911

PSU is connected with a Voltage meassured on the Pi Pin Outputs minimum 5.1Volts

NAS over Wifi

Camera Freezes with the Timestamp on the Picture. After changing the Resolution and click apply mostly it fix the Issue for now.

Resolution is 1600x1200
Omx h264
and Motiondetection by editable mask is setup 
Maybe its something like #1730 again?

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4729666/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4729667/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4729669/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4729671/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/4729672/motion.log)

elmolinger, 2020-06-04:
I thinks its related to 
`Jun  4 14:20:34 meye-steinel user.err motion: [0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout. Trying to do a graceful restart
Jun  4 14:20:44 meye-steinel user.err motion: [0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout did NOT restart, killing it!
Jun  4 14:21:32 meye-steinel user.err motion: [0:motion] [ERR] [ENC] ffmpeg_flush_codec: Error entering draining mode:End of file
Jun  4 14:21:32 meye-steinel user.err motion: [0:motion] [ERR] [ENC] ffmpeg_close: Error flushing codec
Jun  4 14:21:39 meye-steinel user.notice motion: [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
Jun  4 14:21:39 meye-steinel user.notice motion: [0:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
Jun  4 14:21:39 meye-steinel user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Logging to syslog
Jun  4 14:21:39 meye-steinel user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Motion 4.2.2 Started
Jun  4 14:21:39 meye-steinel user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
Jun  4 14:21:39 meye-steinel user.notice motion: [0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)`

holzfelix, 2020-06-09:
> Same Problem here
>
> MotioneyeOS 20190911
>
> PSU is connected with a Voltage meassured on the Pi Pin Outputs minimum 5.1Volts
>
> NAS over Wifi
>
> Camera Freezes with the Timestamp on the Picture. After [...]

Did you have fullHD before? How many cameras? 

Every two oder three days it freezes ..... Where can I find the relevant Logs?

elmolinger, 2020-06-09:
> Did you have fullHD before? How many cameras?

I have only One Cam per Rpi Connected. Original Raspberry Camera Module.

 
> Every two oder three days it freezes ..... Where can I find the relevant Logs?

Teh Log files on MotioneyeOs are in the WebUI > ExpertSettings
[image]

In the meantime, I have followed an unconventional approach.
I installed Raspbian (or Raspberry OS) and set up Motioneye, Motion and ffmpeg with the latest versions. As a result, the problems have now disappeared.

For ffmpeg I used the instructions from @jasaw. https://github.com/ccrisan/motioneye/issues/930#issuecomment-522848452

For Motioneye and Motion I went to https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian, only replacing Motion with the latest version.

My 2 Raspberry's have been running without interruption for 4 days and are busy recording. All error messages have disappeared.

Hope that helps one or the other.

Best regards

hsm-feup, 2020-06-09:
> In the meantime, I have followed an unconventional approach.
> I installed Raspbian (or Raspberry OS) and set up Motioneye, Motion and ffmpeg with the latest versions. As a result, the problems have now disappeared.
> 
> For ffmpeg I used the instructions from @jasaw. [ccrisan/motioneye#930 (comment)](https://github.com/ccrisan/motioneye/issues/930#issuecomment-522848452)
> 
> For Motioneye and Motion I went to https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian, only replacing Motion with the latest version.
> 
> My 2 Raspberry's have been running without interruption for 4 days and are busy recording. All error messages have disappeared.
> 
> Hope that helps one or the other.
> 
> Best regards

At last a light at the end of the tunnel!
I hope this can finally solve the problem!
Thanks

holzfelix, 2020-06-12:
I changed the resolution but same issue...

holzfelix, 2020-06-12:
ah no sorry it looks good. my fail

closed 2020-06-12 by holzfelix