motioneye-project/motioneyeos #2393: Use of BME280 I2C sensors with Motion Eye OS; Overlay onto video feed

achesloc, 2020-05-24:
I did not find exactly what I was trying to accomplish anywhere else, and I *think* this is a good place to place this and then close it.  Of course, if not, please delete or move my input somewhere else.  As for any on-going help, I barely got this to work, so I am not sure I will be of any real assistance.  I just thought this would be useful information for others.

My goal was to have temperature and humidity displayed on top of a live video feed in a non fast network camera setup with Motion Eye OS.  After some trial and error, I landed on using an I2C BME280 sensor.  I used these: https://www.amazon.com/gp/product/B07KR24P6P/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1.  As for the python module / driver to work with the sensor I used the one here:  https://www.raspberrypi-spy.co.uk/2016/07/using-bme280-i2c-temperature-pressure-sensor-in-python/.  That link also has some good information on how to confirm i2ctools are working.  i2ctools are enabled and work in Motion Eye OS.

The first task was to create a script to capture the sensor data.  This is the script I wrote with the bme280.py module in the /data/etc directory and the script in the same directory:

import bme280
 
(chip_id, chip_version) = bme280.readBME280ID()
 
temperature,pressure,humidity = bme280.readBME280All()
 
file_temp = open("/data/etc/TemperatureBME.txt","w");
file_temp.write('{:3.2f}'.format(((temperature*1.8)+32) / 1.))
file_temp.close()

file_humid = open("/data/etc/HumidityBME.txt","w");
file_humid.write('{:3.2f}'.format(humidity / 1.))
file_humid.close()

The above corrects for F temperature read outs.  This the crontab entry, which can be accessed using command: crontab -e
*/2 * * * * python /data/etc/CameraDirectTemperatureBME.py

So, the above script runs every 2 minutes and updates those two text files that live in /data/etc

The next part was to get the video overlay.  First I named the camera, and made sure that stuck using the web MotionEye UI.  I then learned from this post (https://github.com/ccrisan/motioneye/issues/240) that you could alter the text on the left of the live video feed directly, and needed to know the correct port to do so.  _See_ post by maglub commented on Jul 4, 2016.  I did this because I also learned that monitoring output as discussed here is limited to one line (https://github.com/ccrisan/motioneye/wiki/Monitoring-Commands).  So, with that information, I used the monitor_1 scripting capability to do my own overlay as follows, which assume the existence of the two files mentioned above in the /data/etc directory:

#!/bin/bash

outputTemp=$(cat /data/etc/TemperatureBME.txt)
outputHumid=$(cat /data/etc/HumidityBME.txt)
curl --silent http://localhost:7999/1/config/set?text_left="GreatRoom\nTemperature: $outputTemp\nHumidity:    $outputHumid" >> /dev/null

Here is what the overlay looks like for the above right now:

[image]

Lastly, I am also pulling these temperature and humidity values and inserting them into a database since I have deployed a whole network of zero nodes in my house to monitor temperature and humidity on the heals of an HVAC renovation.  These are the curl cron jobs I am using for that:

*/10 * * * * curl -k "sftp://<nameOfDevice>/data/etc/TemperatureBME.txt" --user "<user>:<password>" -o "<location>/TemperatureBME.txt" --ftp-create-dirs
*/10 * * * * curl -k "sftp://<nameOfDevice>/data/etc/HumidityBME.txt" --user "<user>:<password>" -o "<location>HumidityBME.txt" --ftp-create-dirs

I am hoping this would help anybody else looking to do the same thing.  If anybody has any comments on what I did and why it is good/bad, that would be great.  For now, I will post this and close it.

achesloc, 2020-05-24:
As a clarification, the above is with a pi zero w.  Works great.

closed 2020-05-24 by achesloc

starbasessd, 2020-05-24:
> post (ccrisan/motioneye#240
> */10 * * * * curl -k "sftp:///data/etc/TemperatureBME.txt" --user ":" -o
> */10 * * * * curl -k "sftp:///data/etc/HumidityBME.txt" --user ":" -o

First of all, THANK YOU for your time and effort!
I am very much a fan of the bm(x)280 series of sensors. I have numerous
ones around my home, too.
This looks to be well documented, and I look forward to integrating it into
my projects.
Again, thank you.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

achesloc, 2020-05-24:
Sure thing.  I am glad it will be of some use.  It has been ages since computer engineering graduate school and I mostly left the field (I am a patent lawyer now).  The syntax has been the hardest thing.  Conceptually this is all quite easy, it is just difficult to get the right syntax going.  Forums like these are precisely how I glued all this together.  The values I am getting are basically what I expected, and it is nice being able to monitor humidity down here in the basement.

lanche11, 2021-01-06:
[image]
I have 5 Zeros running with the BME280. This works great for on screen values except for one which lost the humidity reading. I have tried searching through the code  for errors and so far haven't found any. Running python bme280.py produces all normal readings. I have also added the barometric pressure which I haven't had issues with.. Any ideas?

achesloc, 2021-01-06:
I will bet the most likely thing is there is something going on with the file permissions/file system issues on the text file the scripts are writing to, if you are using my solution.  Are you confident the one text file is being updated for humidity?

lanche11, 2021-01-06:
Must have had your ear to the rail. What a quick response and a correct analyses. When I finally got the first working I copied to the others and somehow the Humidity ended up when created being different. I have tried removing Humidity and let create a new file. That did not work. How can I switch permission  on the file or am I seeing it wrong?     
     0 -rw-------    1 root     root       0 Jan  6 13:46 HumidityBME.txt
     4 -rw-------    1 root     root       5 Jan  3 18:59 PressureBME.txt
     4 -rw-------    1 root     root       5 Jan  6 13:46TemperatureBME.txt

achesloc, 2021-01-06:
Those permissions match up with what I have on mine for the humidity and temperature text files (I just checked).  Looks like root read/write where root is the owner.  And I think all these scripts are being run as root anyhow.  From what you have above, it looks like both Humidity and Temperature were opened and written to but for some reason the output to Humidity is blank (0 bytes vs. 5 for Temperature).  Note they both show 13:46 January 6 as their most recent modification date as outputted by the ls command.

I would create a debug version of CameraDirectTemperatureBME.py and have it print the Humidity variable to the screen and see if somehow that is blank.  Perhaps add this (in bold) before the file write command.

**print('{:3.2f}'.format(humidity / 1.))**
file_humid.write('{:3.2f}'.format(humidity / 1.))

lanche11, 2021-01-06:
[image]
I found my problem. Out of disk space. I had about 6 large GB mp4s and a 16gb card. I really appreciate your time. 

[root@meye-ba27dd12 etc]# python Camer*
Traceback (most recent call last):
File "CameraDirectTemperatureBME.py", line 13, in <module>
file_humid.close()
IOError: [Errno 28] **No space left on device**

achesloc, 2021-01-08:
Ah. My bad.  I should have flagged that.  I had the exact same problem with mine.  Doh! 

 

Glad you got it fixed!

 

 <https://user-images.githubusercontent.com/46089832/103813121-95171380-502d-11eb-9c3c-90622927b234.png>

on2aav, 2021-02-13:
Just wanted to say, Thanks for a great job !, I modified it for 2 BME280 , getting data on the video !
[image]

Take care !
Luke