motioneye-project/motioneyeos #2395: Upgrade failed and post new install not recognizing my 5 TB External USB drive

BalajiNJL, 2020-05-26:
Hi,

I was running motioneyeos-raspberrypi3-dev20181008 with 4 cameras and one external 5 TB USB HDD for last two years without any issue. My setup is very similar to setup mentioned in this blog
https://randomnerdtutorials.com/cctv-raspberry-pi-based-system-storage-motioneyeos/

Yesterday I decided to upgrade my system. Before upgrade, I backed up my config files via webui. And then I clicked upgrade from webui. It tried to upgrade for about 10 mins and i got 'upgrade failed' message. System rebooted and it failed to come up. I connected my external monitor and noticed system was in a continuous reboot mode. 

So I took the SD card out, re-flashed with new version motioneyeos-raspberrypi3-20190911 and it booted up fine. Tried to import settings using my backed up config file, other than camera name, all my settings were lost. I started re-configuring  manually, I reconnected my 5 TB external HDD, and it is not recognizing external drive at all. I noticed all my data in my drive was lost, then I decided to format it (NTFS) and plugged it back again still no luck.

Same external USB HDD was working earlier (with previous version)  without any issues

Attaching dmesg.log and motioneye.log files, 
[motioneyeOS_dmesg.log](https://github.com/ccrisan/motioneyeos/files/4685522/motioneyeOS_dmesg.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4685536/motioneye.log)

Thanks for your help.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi3-20190911

### Board Model

I am using the following board/model:  Raspberry PI 3B+

### Camera

I am using the following type of camera: RaspberryPI camera

### Network Connection

Wired network connection

### Peripherals

5 TG Seagate USB External HDD

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-05-27:
If you look at the release notes for several versions (
https://github.com/ccrisan/motioneyeos/releases fo 200190427) there have
been major changes and upgrades, and upgrading wouldn't work.
along with that, the structure changed.
With a clean install your 5TB drive should be in /data/media now.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BalajiNJL, 2020-05-27:
Thank you for your response, but I did a clean install and still I don't see my 5 TB HDD. Here is o/p of 'df -h'
[root@videohub media]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               213.0M    197.1M    903.0K 100% /
devtmpfs                369.3M         0    369.3M   0% /dev
tmpfs                   373.8M     16.0K    373.8M   0% /dev/shm
tmpfs                   373.8M      2.4M    371.5M   1% /tmp
/dev/mmcblk0p1           29.9M     14.1M     15.8M  47% /boot
/dev/mmcblk0p3           57.3G     54.6M     54.3G   0% /data
/dev/mmcblk0p3           57.3G     54.6M     54.3G   0% /home/ftp/sdcard
/dev/mmcblk0p3           57.3G     54.6M     54.3G   0% /home/ftp/storage
[root@videohub media]#

starbasessd, 2020-05-27:
Power off
make sure the 5TB drive is plugged in
power on the PI
SSH into the PI
type in:
dmesg | grep sda
what are the results?
If it doesn't show, try the drive on a computer. Does it show up there?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BalajiNJL, 2020-05-27:
I took the HDD and plugged it onto my Ubuntu laptop, it recognized without any issues. I still went ahead and reformatted the drive with Ext4 just to make sure. 

Plugged it back on RaspPI, powered it down and up again, still no luck.

[root@videohub ~]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               213.0M    197.1M    903.0K 100% /
devtmpfs                369.3M         0    369.3M   0% /dev
tmpfs                   373.8M     16.0K    373.8M   0% /dev/shm
tmpfs                   373.8M      2.4M    371.5M   1% /tmp
/dev/mmcblk0p1           29.9M     14.1M     15.8M  47% /boot
/dev/mmcblk0p3           57.3G     54.7M     54.3G   0% /data
/dev/mmcblk0p3           57.3G     54.7M     54.3G   0% /home/ftp/sdcard
/dev/mmcblk0p3           57.3G     54.7M     54.3G   0% /home/ftp/storage
[root@videohub ~]# /dev/sda
-sh: /dev/sda: No such file or directory
[root@videohub ~]# dmesg | grep sda
[    4.166196] sd 0:0:0:0: [sda] Spinning up disk...
[   89.546443] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[   89.599451] sd 0:0:0:0: [sda] 9767541167 512-byte logical blocks: (5.00 TB/4.55 TiB)
[   89.599468] sd 0:0:0:0: [sda] 2048-byte physical blocks
[   96.733905] sd 0:0:0:0: [sda] Test WP failed, assume Write Enabled
[  103.842672] sd 0:0:0:0: [sda] Asking for cache data failed
[  103.842689] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  104.264694] sd 0:0:0:0: [sda] Spinning up disk...
[  114.481991] sd 0:0:0:0: [sda] Read Capacity(16) failed: Result: hostbyte=0x01 driverbyte=0x00
[  114.482003] sd 0:0:0:0: [sda] Sense not available.
[  114.482076] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x01 driverbyte=0x00
[  114.482086] sd 0:0:0:0: [sda] Sense not available.
[  114.482106] sd 0:0:0:0: [sda] 0 512-byte logical blocks: (0 B/0 B)
[  114.482117] sd 0:0:0:0: [sda] 2048-byte physical blocks
[  114.482155] sd 0:0:0:0: [sda] Write Protect is on
[  114.482167] sd 0:0:0:0: [sda] Mode Sense: 59 e6 93 1e
[  114.482205] sda: detected capacity change from 5000981077504 to 0
[  114.483139] sd 0:0:0:0: [sda] Read Capacity(16) failed: Result: hostbyte=0x01 driverbyte=0x00
[  114.483151] sd 0:0:0:0: [sda] Sense not available.
[  114.483224] sd 0:0:0:0: [sda] Read Capacity(10) failed: Result: hostbyte=0x01 driverbyte=0x00
[  114.483234] sd 0:0:0:0: [sda] Sense not available.
[  114.483334] sd 0:0:0:0: [sda] Write Protect is off
[  114.483346] sd 0:0:0:0: [sda] Mode Sense: b3 b9 5c fa
[  114.483386] sd 0:0:0:0: [sda] Attached SCSI disk
[  114.503431] udevd[1430]: inotify_add_watch(6, /dev/sda, 10) failed: No such file or directory
[  114.513570] udevd[1430]: inotify_add_watch(6, /dev/sda, 10) failed: No such file or directory
[root@videohub ~]#

starbasessd, 2020-05-28:
Try the following commands:
mkdir /data/meda/sda1
mount /dev/sda1 /data/media/sda1
df -h

Let me know the results

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BalajiNJL, 2020-05-28:
but there is no dev/sda1, as per dmesg above it is not recognizing /dev/sda.

[root@videohub ~]# mkdir /data/media/sda1
[root@videohub ~]# mount /dev/sda1 /data/media/sda1
mount: mounting /dev/sda1 on /data/media/sda1 failed: No such file or directory
[root@videohub ~]#

starbasessd, 2020-05-29:
Try formatting the USB drive with 2 partitions, 1 3.5TB 11.5TB formatted as
you want., then try to connect it to the PI
It never fully spins up.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?