motioneye-project/motioneyeos #2397: Webinterface Bootstrap

weiserhei, 2020-05-27:
Hello,
this issue is not targeted at motioneyeos in particular, rather than the webinterface which is the same for motioneye.

Whats the reason behind building it without any frontend library? While the hand-crafted UI Elements and colors have their charm, Im having some trouble using it on my mobile device. 

Beeing curious I came up with a working prototype for the Settings Panel using Bootstrap. I havent looked at all the Pages/Elements yet, but it looks promising to me. If there is any interest in "bootstrapping" motioneye I might give this a try. Maybe there are some obstacles hidden which renders all of the work useless, so please let me know. 

<details>
  <summary>Screenshots</summary>

[image]

[image]

[image]
</details>

jasaw, 2020-05-27:
If you are not aware, https://github.com/ccrisan/motioneye is the web UI part of MotionEyeOS. Can you please repost this question there? I think you might get more responses there.
Personally, I like bootstrap.

weiserhei, 2020-05-27:
Thanks for the headsup, I close this one then 👌

closed 2020-05-27 by weiserhei