motioneye-project/motioneyeos #2398: Stream freezes when trying to access it through Home Assistant

MichielvanBeers, 2020-05-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 

motionEye Version | 0.41
Motion Version | 4.2.2
OS Version | motionEyeOS 20190911

### Board Model

I am using the following board/model: Raspberry Pi 4, 2GB RAM

### Camera

I am using the following type of camera: V4L2 Camera connected through USB

My camera model is: USB 720P Webcam ([link](https://nl.aliexpress.com/item/33004665726.html?spm=a2g0o.productlist.0.0.3cb05da6c47fP1&algo_pvid=849f384a-d875-45ed-bd04-7c11f489f28d&algo_expid=849f384a-d875-45ed-bd04-7c11f489f28d-12&btsid=0ab50f4915906651651767455e5ab4&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_))

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi in a seperate vlan.

### Log Files

I consider the following log files relevant to this issue:

[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/4695206/motioneye.1.log)
I've attached the debug log for further information.

[boot.log](https://github.com/ccrisan/motioneyeos/files/4695220/boot.log)
 
### Error

My camera freezes when I try to access the camera through the [Home Assistant Image Processing component](https://www.home-assistant.io/integrations/image_processing/). The log gives me the `ERROR: mjpg client timed out receiving data for camera 1 on port 8081`. Rebooting the system using the web interface doesn't always restore the issue. Power cycling always fixes the issue. (NB: the image froze at 2020-05-28 13:17:38)

Things I have tried to solve the issue are (without solving the issue):

1. Reducing the framerate to 10fps
2. Disabling SSH
3. Disabling Samba 

Thank you very much for your help! Please let me know if I can provide any other information,

Warm regards

starbasessd, 2020-05-28:
Have you checked the CPU temp and voltage of the power supply?
log into Pi via ssh
CPU Temp command
vcgencmd measure_temp

To see if your Pi has throttled due to low voltage
vcgencmd get_throttled

Let me know the results, especially if the camera has frozen before running
the commands.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

MichielvanBeers, 2020-05-28:
@starbasessd , thank you for the quick response. Herewith the results:

Outcome vcgencmd measure_temp: `temp=34.0'C`
Outcome vcgencmd get_throttled: `throttled=0x50000`

I've been also experimenting some more, and it looks like the screen only freezes when my doorbell rings at the same time. Some background info: the camera is at my front door and is triggered when the doorbell is rang. When I disable the ring of the doorbell, the image doesn't freeze. The Pi4 is plugged in in the adjacent socket. Could this by causing an issue?

Finally: when the screen freezes, the load information at the bottom of the view continues to change values, though the image doesn't change (see picture). Not sure if that tells you anything, but maybe it helps..
[image]

Thanks anyway!

starbasessd, 2020-05-28:
the voltage is at least momentarily dropping, probably causing the freezing.
Is the doorbell connected directly to the PI? Where is the PI getting it's
power? The Camera may also be borderline pulling too much power from the PI.
I would run the camera to a Self-powered USB Hub then to the PI as a start.
I would run the PI itself from a separate power source (don't use the
powered hub to owere the PI too.
If the doorbell is connected directly to the PI's GPIO I'd work on a buffer
circuit, or at least put a drop resistor in there (either to VCC or GND, if
it currently just connects the GPIO ping to VCC or GND (4.7K usually works)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

MichielvanBeers, 2020-05-28:
The Pi is powered through a Huawei charger that could output max 4,5 - 5A (see picture). The doorbell is connected to the gpio pins of a Sonoff flashed with Tasmota. Both are separately powered.
[image]

I don't have a self-powered USB hub at hand, but I'll try to get my hands on one. I'll update you as soon as I have more information. Thanks for tips!

starbasessd, 2020-05-28:
> on one. I'll update you as soon as I have more information. Thanks for the

I believe your power supply for the PI is too small. It only puts out 2A
standard. It would put out the 4.6A to a device that can trigger it, like,
say a Galaxy Tablet or Phone, that has the correct circuitry for 'fast
charge. The Pi doesn't have that circuitry, so it would only pull the 2A.
The PI 4 has issues with some USB C cables, too, but I think your power
issue is the fastCHARGER you are using. I'd get an 'Official' power supply,
or at least one certified by AdaFruit, PiMoroni, or other reputable dealer,
they're $7-$10 US

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

MichielvanBeers, 2020-05-28:
I've directly ordered the official power supply :). Expected delivery is on the 1st of June, so I'll keep you posted!

starbasessd, 2020-05-28:
Please and thank you. I have an 'Official' and a certified by adafruit,
both 3A USB C...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

MichielvanBeers, 2020-05-30:
The power supply came in! But it unfortunately didn't solve the issue.. I've also tried to fix the issue using by adding a [pull up resistor to the Sonoff](https://github.com/arendst/Tasmota/issues/1003) and adding more shielding, but both didn't work sadly enough.

I think my last resort is just to power cycle the device when the doorbell has rang. Not ideal, but with a simply automation that is fixed.

I think this issue can then by closed, since this is hardware issue and not an issue with MotionEye. 

Thanks a lot @starbasessd for the effort and thinking along with me!

closed 2020-06-23 by MichielvanBeers

IanEatsBabies, 2022-08-11:
It's a bug in MotionEye.

I had it on an RPi3, 4, and an i5 desktop I converted to HA. Once you reboot MotionEye when multiple cameras attached it loses the camera assignments and just assigns camera's at random (this may be Linux though). After that HA will always hang while viewing the camera.

Debating wheter I'll run through everything or just ditch that pos sofware MotionEye and replace it with something better. 
Right pissed about ME destroying yet another HA install :@