motioneye-project/motioneyeos #2399: Can I flash MotionEyeOs on an old android device

JakubKoralewski, 2020-05-29:
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

Can I flash MotionEyeOs on an old android device? I'm not talking about a client, but actually running the OS, being able to work as a hub for other devices on the network?

Thesinofthetin, 2020-05-30:
I dont think so.  But maybe in the future

starbasessd, 2020-05-30:
If you think having 14 different versions (4 different PI versions) is bad,
because of the hardware differences of all the SBCs supported, it would
rapidly go to thousands needed for android devices. If you know an android
programmer, they might be able to take the source code for motionEye (not
OS) and compile it as an app...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JakubKoralewski, 2020-05-30:
> If you think having 14 different versions (4 different PI versions) is bad, because of the hardware differences of all the SBCs supported, it would rapidly go to thousands needed for android [...]

@starbasessd You're right. Admittedly, I only want to use motioneyeos as a hub, so camera support being unnecessary for me, getting stuff like the internet modem, touch screen support, GPU, h/w codecs etc. would be necessary for this endeavor to make sense. 

That said, how about using something like [the Linux Deploy apk](https://github.com/meefik/linuxdeploy)?

starbasessd, 2020-05-30:
" That said, how about using something like the Linux Deploy apk
<https://github.com/meefik/linuxdeploy>?"
That's where you need the skills of a developer or programmer in android
(or iOS). I have no experience with that process, I muddle along, ask my
gurus, and try to do tech support here....

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JakubKoralewski, 2020-05-30:
From Linux Deploy README:
> Bootstrap: Alpine, Arch, CentOS, Debian, Fedora, Kali, Slackware, Ubuntu, Docker or from rootfs.tar

Could MotionEyeOs be put into a Docker container or `rootfs.tar` and Linux Deploy be reused instead of writing a new app specifically for moitioneyeos?

starbasessd, 2020-05-30:
Maybe? You can try for yourself. The Buildroot source
https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source builds a
rootfs/bootfs file system partitions images (total combined size aof almost
340M) during the build phase, you might be able to do something with those?
This is an opensource project, and the source code is available at:
https://github.com/ccrisan/motioneyeos/archive/nightly-dev.zip

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JakubKoralewski, 2020-05-30:
Unfortunately my S3 is too old, (Linux Deploy's min SDK is 21 which is Android 5.0), so it wouldn't run Linux Deploy on it's original 4.x version. I would have to flash the most recent LineageOS image before S3 support was discontinued.

And also I don't know if I'd be able to do this, as my understanding of Linux is very basic :confused:

starbasessd, 2020-05-30:
If the phone is that old, does it really matter if you brick it? I am
always prepared to re-image / re-install / etc with stuff I know there's a
risk, but it's called hands-on learning.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JakubKoralewski, 2020-05-30:
I was mostly talking about the time required to flash a new system + I'm a bit hesitant about bricking still since this old and trusty S3 has saved my a couple times when other phones failed 😄 

That said, bricking shouldn't happen I flashed LineageOS a couple times, and Linux Deploy's mechanism (if I understand correctly) is reversible, so therefore also safe I guess

closed 2022-01-19 by starbasessd