motioneye-project/motioneyeos #2400: Raspberry PI Zero + Night Vision Issue

STUKguy, 2020-05-29:
Hi,  basically i brought the below
https://www.ebay.co.uk/itm/5-Megapixel-Infrared-Night-Vision-IR-CUT-Camera-For-Raspberry-Pi-4B-3B-Zero-NEW/293554729746?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

Connected it and when i power on the PI Zero W i get this
[image]

And this is what i get when i log in to motion eye os (latest build) bare in mind it is the middle of the day and room i bright
[image]

This is with CSI Camera led off

If i adjust the small screw next to IR thing on the side
[image]

so they are both off 
[image]

i still get this when i log into motion eye
[image]

If i enable CSI and reboot the camera i still get the same thing but when i face the camera towards my PC monitor the colour is fine but when i face the camera around my room the colour is all purple-ish
[image]
[image]

and finnaly if i turn the sreaws again nothing changes.

I have played about ith the colour setting on Motion eye but this does not and like i said when i point it towards my screen all the colours are correct.

Any ideas, sorry about the really long post

JohanD78, 2020-05-29:
Your camera is a no ir-filter type (ir cut)

What you see is normal for a no-ir camera.

STUKguy, 2020-05-29:
@JohanD78 thank you for your reply, the problem i am having as explained above in a bright room the image is still the same but from my understanding this should be happening in a bright room.

JohanD78, 2020-05-29:
[image]

[https://raspi.tv/2013/pinoir-whats-it-for-comparison-of-raspicam-and-pi-noir-output-in-daylight](url)

Dont matter if it is dark or bright. You got a no-ir cam and thats the tint you get with those.

STUKguy, 2020-05-30:
@JohanD78 Damn thank you that explains a lot. Im trying to build a out door camera with night vision what would suggest i use for the camera?

omgitsaheadcrab, 2020-05-30:
The camera you have is the only one that will be able to see anything in the dark, so stick with that.

zubzero689, 2020-06-01:
i was doing similar you might want to get more IR light i had to i couldnt see a thing unless i was within like 10 feet of the camera and i hung them under my camera different power but works great and most are rated to 80 feet

PsyEng, 2020-06-06:
@STUKguy What you're searching, is something like this:
https://www.amazon.de/dp/B07XTBQ1BN/ref=cm_sw_r_other_apa_i_My-2Eb5DCGWCG
This has a built-in IR-cut filter, which can be activated and disabled automatic, so you've a clean/real view on day and nightvision at night.

PsyEng, 2020-06-06:
@STUKguy What you're searching, is something like this:
https://www.amazon.de/dp/B07XTBQ1BN
This has a built-in IR-cut filter, which can be activated and disabled automatic, so you've a clean/real view on day and nightvision at night.

STUKguy, 2020-06-06:
> @STUKguy What you're searching, is something like this:
> https://www.amazon.de/dp/B07XTBQ1BN
> This has a built-in IR-cut filter, which can be activated and disabled automatic, so you've a clean/real view on day and nightvision at night.

Thank you so much, I will order this now

PsyEng, 2020-06-07:
@STUKguy You´re welcome. I find another way, but accident today^^
You can set under the "Extra Motion Options” box this parameter and it should be fixed
mmalcam_control_params -awb greyworld

STUKguy, 2020-06-07:
@PsyEng Hi, Thank you agian, i tried that but nothing happened
[image]

STUKguy, 2020-06-07:
@PsyEng its working now kind of
[image]
Atleast nothing is pinky/purple. I have ordered the camera you suggested just waiting for that to arrive before i decide which one i want to use. Thank you again soo much

STUKguy, 2020-06-28:
@PsyEng Thank you again for your help the camera arrived last week (or the week before) and i finnaly have set everything up and it works great had to make some tweaks (used Pi insted of Pi Zero) Ive just noticed the camera is slightly out of focus, is there anyway to adjust this with out dismantling the camera?

PsyEng, 2020-06-29:
@STUKguy You can turn the lense to the right and left to steer focus. It could be a little bit hard, so don't worry. The focus is quit fine, so you've to be patient to get the perfect spot.

PsyEng, 2020-06-29:
@STUKguy Two things. First, ist not possible to view trough a window, because the leds blends), second, you didn't input the code for greyworld or the leds blends the photodiode so the camera doesn't get in nightvision mode. It looks like the camera doesn't get in nightvision, because I can see violet lights on the image.

STUKguy, 2020-06-29:
@PsyEng the camera has been placed inside a camera enclosure and is behind a 1mm thick transparent plastic sheet. I didnt think the camera
[image]
needed the code but i will put it in now and see how it works tonight

PsyEng, 2020-06-29:
Trough glass or something else, you won't be able to see something else. The infrared spots will reflect on the glass and the camera is blinded.

STUKguy, 2020-06-29:
@PsyEng the camera needs to be on the outside of the plastic sheet? So if I cut a hole that should fix the problem?

PsyEng, 2020-06-29:
Maybe. I don't know exactly, if the infrared spots can get through the plastic. You've to test it by yourself.

STUKguy, 2020-06-29:
@PsyEng what I will do is see how the code works tonight with the camera out side if no change the I will bring in side and test again

zubzero689, 2020-06-29:
Just an FYI when I put my cameras out side I had to put out some IR leds to get a better view i bought them of Amazon because I put mine in housing of dummy cameras and could only use 1 led of the two it comes with

STUKguy, 2020-07-10:
@PsyEng So I had to do some more changes I had to upgrade to a PI 4B 2GB found out that the LED need to be on the outside but the night sensor can be behind a glass to work correctly which works well but and images beyond 3 meters come out very blurry. The spec sheet for the IR LED's online 3 meters is the maximum length for a decent image these LEDS are rated at 3w. Based on actual CCTV camera with Night vision they usually come with a 12v IR led which from previous cameras I know can get a decent image further that 3 meters just unsure how to implement this in to my project as I doubt the PI can power a 12v LED

STUKguy, 2020-07-10:
@zubzero689 Thank you for your comment I also had to use 1 LED as I could not get the 2nd to power unsure if this was a faulty LED.

jasaw, 2020-07-10:
@STUKguy It's much easier to run a 12V power supply, run the IR LEDs off the 12V, and step the 12V down to 5V for the RPi. You can easily find a step-down DC-DC converter for $2 for the 12V to 5V conversion.

STUKguy, 2020-07-20:
@PsyEng @jasaw Thank you for you help, but i have had to give up on this project as it is starting to cost to much and is just becoming really annoying

So this is the layout i managed
[image]

i re routed the night sensor so it would work with the 12v IR LED ring but then come to the night i got this image when i pointed the camera outside
[image]

Even after turning Greyscale off same thing

I also pointed the camera indoors
[image]

This is the front veiw of the camera
[image]

i tried with the NOIR camera and excatly the same no differance, i am at a loss how to fix this but i am assumeing the issue im having now is with the IR Leds being to close to the lens but as this project has cost me to much already and i have found the Reolink 420 camera has what i need i will abandon this project and purchase the Reolink and re purpose the PI 4 for a NAS stoarge im looking to build.

Thank you again.

jasaw, 2020-07-20:
@STUKguy I understand your frustration. I built my own cameras and had the same issues where the ir light was drowning the camera because of ir reflection. I ended up 3d printing a shield around the camera, added foam tape to press this shield against the housing glass to make sure there's no ir leakage. I also 3d printed mounting brackets to hold all the parts in place. I even designed my own ir led board ring around the camera. After 2 years in the outdoor, all the cameras are still working well while the Chinese cheap counterparts are partially broken by now.

In the future, I would just install off the shelf good quality cameras with good optics. Don't bother with cheap cameras.

STUKguy, 2020-07-20:
@jasaw I didnt buy a cheap chinise camera, by the looks of it mine and your idea were the same i just dont have access to a 3d printer. Looking further at your instruction i wish i found that out a lot earlier it would have made my life a lot easier

closed 2020-08-14 by STUKguy