motioneye-project/motioneyeos #2404: Power off button (on GPIO)

AN4LOG2K, 2020-05-31:
Is there a way to add a physical button (such as a momentary switch connected to GPIO 3 or a combination such as GPIO 21 & ground) to power off safely both motioneyeos & the rasp pi. I am using a pi zero W / dev20200223 / pi cam v2 . 

The reset (RUN) button could be used to switch the whole system up again. 

The goal is to make a portable camera that can be activated and deactivated with the push of a switch when needed without the need for SSH or a Browser. 

Cheers

starbasessd, 2020-05-31:
Yes. There is a GPIO pin and a dtoverlay file that can be copied into the
/boot partition and a couple of lines into config.txt
There is a good set of instructions at
https://forums.balena.io/t/dtoverlay-gpio-shutdown-and-gpio-poweroff-options-not-working/36467

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

AN4LOG2K, 2020-06-07:
Hi there thank you for your answer. I haven't had the chance to work on this project yet unfortunately. But I will get back to you once I have successfully managed to do it.