motioneye-project/motioneyeos #2406: No Network

bbuehrer, 2020-06-02:
I have a Raspberry PI with MotionEye OS.

Is there a way to operate the Raspberry PI with MotionEye without a network connection?

I want to record videos outside the house and then connect to the network later in the house to watch the videos.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (20190911).

### Board Model

I am using the following board/model: (Raspberry PI 3B).

### Network Connection

My motionEyeOS unit is connected to the network via: (Ethernet).

starbasessd, 2020-06-02:
Yest. I've attached a set of instructions and config files to show you how.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bbuehrer, 2020-06-02:
Sorry, i can't see any attachment in your post.

starbasessd, 2020-06-02:
[ConfigFiles.zip](https://github.com/ccrisan/motioneyeos/files/4717062/ConfigFiles.zip)

Here you go. I keep forgetting the system strips attachments from emailed responses.

bbuehrer, 2020-06-02:
Hi,

is there no way to start the Raspberry without a network connection and next time with a network connection and it recognizes this automatically?

Example:
OS_NETWORKLESS = "auto"

starbasessd, 2020-06-02:
No, not at this time.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?