motioneye-project/motioneyeos #2407: is there a version for the Banana Pi BPI-M2 ZERO?

flexizon, 2020-06-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

starbasessd, 2020-06-03:
Have you tried the bananapim1 image? If not, you could try to 'roll your
own' until there is. Instructions here:
https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source
You could see if motioneye can run on whatever linux OS your board runs.
(Like motionEye on Raspbian for the PI)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

flexizon, 2020-06-04:
I tried the bananapim1 image but it doesnt boot up it stays dead.

i try to make an image on my raspberry pi for my bananapi but i cant figure out how to do exacly.
i dont know if its going to work anyway. i am a little bit a noob in this.

I am running rasbian on my raspberry .
the first steps like downloading the image (step 2) is ok.
But know i am trying to understand step 3 step 4 and 5.

is it possible to write in  ./build.sh [board] xconfig just like this
 ./build.sh bpi-m2 xconfig?

kind regards..

sagitt, 2021-04-12:
i want it too :(

911pcdoc, 2021-08-23:
also waiting for an image for this board!

d-looper, 2021-09-03:
I would like to see those as well. Im curious to see if the BPI-M2 Zero outperforms the pi zero w.

911pcdoc, 2021-09-05:
m2-zero just landed on my desk gonna attempt to build the os for it today and start my testing!

911pcdoc, 2021-09-05:
> m2-zero just landed on my desk gonna attempt to build the os for it today and start my testing!
/Desktop/motioneyeos$ ./build.sh [bananapi-m2-zero] menuconfig
unknown board: [bananapi-m2-zero]

/Desktop/motioneyeos$ ./build.sh bananapi-m2-zero menuconfig
unknown board: bananapi-m2-zero

starbasessd, 2021-09-05:
You need to copy the board folder into the motioneye board build folder tree and may need to generate other files to enable it. Not for the faint of heart. 
Try loading the latest version of available OS (Debian or Ubuntu) and run the install motioneye for that OS on your board to confirm it can be done that way.
https://github.com/ccrisan/motioneye/wiki/Installation

911pcdoc, 2021-09-06:
UPDATE:
so I was able to get armbian installed on the banana pi and ruing through the install I get all the prerequisites installed and when I run pip install motioneye

 Downloading https://files.pythonhosted.org/packages/b3/d0/a20d8440b71adfbf133452d4f6e0fe80de2df7c2578c9b498fb812083383/Pillow-6.2.2.tar.gz (37.8MB)
    99% |████████████████████████████████| 37.8MB 4.7MB/s eta 0:00:01Killed

and that's it I can go no further maybe someone can help

armbian 21.08.1 Buster with Linux 5.10.60-synxi

starbasessd, 2021-09-06:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Step 4.

911pcdoc, 2021-09-06:
> https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
> Step 4.

thank you oh so much!!

911pcdoc, 2021-09-06:
with that, I was able to get motioneye installed on the Banana m2 zero and it does run however I don't think my camera is what it needs as it's off a RASPBERRY PI  so I will order a banana pi cam 

starbasessd THANK YOU SO SO MUCH I will keep you all posted on the outcome of this project! as the price of the raspberry pi zero w has hit almost 40 bucks now and these are faster and cheaper at this point!

d-looper, 2021-09-07:
@911pcdoc so you installed motioneye and not motioneyeos on it?

sagitt, 2021-10-09:
> with that, I was able to get motioneye installed on the Banana m2 zero and it does run however I don't think my camera is what it needs as it's off a RASPBERRY PI so I will order a banana pi cam [...]

support fast camera mode?

starbasessd, 2021-10-09:
Only motionEyeOS supports Fast Network Camera mode, and I do not know if it supports it on other boards, and would only support it if it uses CSI cameras (ribbon cameras).
The utility that does the Fast Network Camera mode is streamEye, not motion or motionEye.
Specifics are here:
https://github.com/ccrisan/motioneyeos/wiki/Fast-Network-Camera
https://github.com/ccrisan/streameye
It is open sourced by CCrisan and uses any /dev/video0 device... Might be usable on other boards running similar linuxs...

sagitt, 2021-10-09:
> Only motionEyeOS supports Fast Network Camera mode, and I do not know if it supports it on other boards, and would only support it if it uses CSI cameras (ribbon cameras). The utility that does the [...]

thanks, i hope in a future support of this board from motioneyeos 👍

starbasessd, 2022-01-19:
No comments after more than 60 days. If you are still having issues, please open a new issue.

closed 2022-01-19 by starbasessd

evandene, 2022-06-26:
On my Banana pi M2 zero I successfuly installed and use MotionEye on Armbian 21.02.1 Buster
Follow these instructions step by step. All command line started with 'sudo'
https://github.com/motioneye-project/motioneye/wiki/%28Install-On-Ubuntu-%2820.04-or-Newer%29
Have fun