motioneye-project/motioneyeos #2409: [Wiki addition] Full Debian Buster installation example

zaggynl, 2020-06-04:
Motioneyeos on Debian Buster

HW used
Logitech C270 USB webcam
https://www.supermicro.com/en/products/system/Mini-ITX/SYS-E200-9B.cfm
Another internet connected PC to download ISO on, prep USB stick, view this guide

OS
https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/
debian-10.4.0-amd64-netinst.iso

Sources
https://github.com/ccrisan/motioneye/wiki/Install-On-Debian

https://wiki.debian.org/WiFi/HowToUse#wpa_supplicant

https://www.matthewcantelon.ca/blog/debian-buster-wifi/

https://wiki.debian.org/rtl819x#Installation

Download ISO, verify shasum and dd to USB stick
download ISO: wget https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/debian-10.4.0-amd64-netinst.iso
Check hash of downloaded file: sha256sum debian-10.4.0-amd64-netinst.iso
Compare hash to contents of https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/SHA256SUMS
Plug in usb stick, ethernet cable, keyboard and display to machine intended to run Debian and Motioneyeos on
Check USB stick device name with dmesg (sdX where X needs to be changed to b or c)
dd the ISO the the usb stick: dd if=debian-10.4.0-amd64-netinst.iso of=/dev/sdX
Install Debian

Select SSH server during install, unselect desktop, for this example guide I’ve not set a root password

Example IP addresses used in this guide:
192.168.1.123 by DHCP on ethernet cable
192.168.1.245 by DHCP on delock wifi dongle (added static lease for convenience)
check by: ip addr | grep inet

Prepare sources list, needs some nonfree bits as per motioneye debian guide
echo “deb http://www.deb-multimedia.org buster main non-free” | sudo tee -a /etc/apt/sources.list
echo “deb http://deb.debian.org/debian/ buster contrib non-free” | sudo tee -a /etc/apt/sources.list sudo apt-get update -oAcquire::AllowInsecureRepositories=true
Install required packages
I’ve added some other packages that are useful, firmware-realtek is specific to the wifi dongle I used, otherwise the dongle wouldn’t get recognized and cause my debian machine to freeze, many errors in dmesg
sudo apt-get install deb-multimedia-keyring wireless-tools wpasupplicant firmware-realtek motion v4l-utils python-pip python-dev python-setuptools curl libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev lm-sensors atop htop fatrace iotop lshw -y
sudo apt update
sudo apt upgrade -y
sudo reboot
connect wireless USB dongle
sudo iwconfig (check if adapter listed and get name, in this example is wlx5cf3704af363)
sudo ip link set wlx5cf3704af363 up (enable interface)
sudo iwlist wlx5cf3704af363 scan | grep -i ESSID (get name of nearby WiFi APs)
setup network
sudo wpa_passphrase “networkname” | sudo tee /etc/wpa_supplicant/wpa_supplicant.conf > /dev/null
(enter password)
test if works:
sudo wpa_supplicant -c /etc/wpa_supplicant/wpa_supplicant.conf -i wlx5cf3704af363
OK if looks like: wlx5cf3704af363: CTRL-EVENT-CONNECTED - Connection to bc:a5:11:31:f8:61 completed [id=0 id_str=]
Ctrl+C to disconnect (takes a while)
sudo cp /lib/systemd/system/wpa_supplicant.service /etc/systemd/system/wpa_supplicant.service
sudo nano /etc/systemd/system/wpa_supplicant.service
change ExecStart line to:
ExecStart=/sbin/wpa_supplicant -u -s -c /etc/wpa_supplicant/wpa_supplicant.conf -i wlx5cf3704af363
comment Alias line
enable service: sudo systemctl enable wpa_supplicant.service
create new service:
sudo systemctl edit --force dhclient.service
add:
[Unit]
Description=DHCP Client
Before=network.target

[Service]
Type=simple
ExecStart=/sbin/dhclient wlx5cf3704af363

[Install]
WantedBy=multi-user.target

enable service: sudo systemctl enable dhclient

still have this, not sure if relevant?
sudo nano /etc/network/interfaces
add:
#wifi plug
allow-hotplug wlx5cf3704af363
iface wlx5cf3704af363 inet dhcp
wpa-ssid networkname
wpa-psk networkpassword

test if wifi works
unplug ethernet and run: sudo reboot
try to connect by WiFi IP address

install motioneyeos
sudo pip install motioneye
sudo mkdir -p /etc/motioneye
sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
sudo mkdir -p /var/lib/motioneye
sudo cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
sudo systemctl daemon-reload
sudo systemctl enable motioneye
sudo systemctl start motioneye

Finish Motioneyeos configuration
Browse to webinterface running at http://192.168.1.245:8765/
default login:
username: admin
leave password blank
configure camera

-set resolution/framerate for camera, make sure framerate can be reached and doesn’t overtax CPU
-tweak motion detection so not smallest movement of tree is recorded
-set time to keep media, depends on amount of recorded movement in day, type and size of -storage media (SD card wears out quickly compared to HDD/SSD)

starbasessd, 2020-06-04:
Was there a question in there somewhere?
If not, please close.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zaggynl, 2020-06-04:
I wanted to make an addition to the wiki but couldn't find a way, hence an issue, I'm sorry if this is not the correct method.

starbasessd, 2020-06-04:
We as users can't modify the wiki. I have a number of 'how -to's I'd like
to see there, also, but...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?