motioneye-project/motioneyeos #2410: Playstation 3 Eye Camera Compatibility Issue with Raspberry Pi Zero W

Wthebinder, 2020-06-05:
I have a weird compatibility issue. My Playstation 3 Eye Camera works with my Raspberry Pi Model 3 B+ but not with my Raspberry Pi Zero W. The Pi Zero recognizes it but it says that it is "Unable to Open Video Device". Even with "Enable CSI Camera LED" on.

Any thoughts about why this is happening?

starbasessd, 2020-06-05:
Have you confirmed functionality with Raspbian?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Wthebinder, 2020-06-05:
Yes. Feel free to check out this forum here: https://www.raspberrypi.org/forums/viewtopic.php?t=148886

Wthebinder, 2020-06-12:
Any thoughts why this would be happening? Can I help troubleshoot more on my end?

starbasessd, 2020-06-12:
Does it work on your Zero with Raspbian?
I know the link says they work on other PI's. Does your USB-OTG adapter
work with something like a memory card?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?