motioneye-project/motioneyeos #2412: wpa_supplicant deleted after modifying any item with the web interface.

ptoulza, 2020-06-06:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi2-20200203.img.xz
### Board Model

I am using the following board/model: Raspberry PI 2

### Camera

I am using the following type of camera: Embedded camera on RASPI2.

### Network Connection

My motionEyeOS unit is connected to the network via: WIFI

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

USB WIFI ADAPTER

### Log Files
Problem : According to the wiki, I've added a working wpa_supplicant.conf to the SD root.
This working great, even after multiple reboot.
BUT... If I modify ANY configuration item (example : Samba property), in any category using web interface, and then reboot. The raspberry reboots in loop, because the wpa_supplicant.conf has been deleted.
I have to manually add again the wpa_supplicant.conf to the sdcard unless it can boot again.
I've tried to modify the /etc/wpa_supplicant.conf to have a permanent config, with a RW mount of the / filesystem. 
But, even /etc/wpa_supplicant was well written, no more chance, after a modification with the web interface, everything is lost.
Do you have any hint ?
Thank you for your great work !

postal302, 2020-06-15:
I also am having the same issue. I'm running a Pi Zero W and MotionEyeOS 20200606.

I changed the camera name, video resolution and frame rate. After rebooting, it will not come back up. The board LED stays on. I checked the SD card and wpa_supplicant.conf is deleted. I copied it to the SD card again, powered up again to the same result. Again the wpa_supplicant.conf is gone.

I'd be happy to provide logs, but I'm not sure how to get it.

ptoulza, 2020-06-15:
I've found an ugly workaround for this problem: I put the wpa_supplicant.conf to the /root path. 
Then I've modified the wifi script in /etc/init.d so it looks for the wpa_supplicant.conf in /root or /boot but no more in /etc/ or /data/etc.
It works well now but, as said.. it's quite ugly.

tomasrojasc, 2020-06-21:
@ptoulza can you explain a bit more, I cannot find the path you are talking about. Apparently I have the same problem