motioneye-project/motioneyeos #2413: Releases versions = beta or stable?

ralfberger, 2020-06-06:
Hi,
I use motionEyeOS 20190911 on a PiZero.
When I click on the Update button I get the answer that there is no newer version. This is also the version that is offered for download.
But what about the realeases version, are they beta or why are they not installed during the update check. 
Best regards

starbasessd, 2020-06-06:
Can't answer the question, however, I use 20200203 for everything around
me, and recommend it to people with most issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JohanD78, 2020-06-06:
1: In the webui you can "Enable Prereleases" under expert settings. That way you can update to latest version.

2: Dont use anything after 20190911 if you use continueus recording or motion recording since those are broken in the later pre-releases. There have been upload activity on the git today so hopefylly there will be new pre-releases that might fix those bugs.

ralfberger, 2020-06-06:
Thanks for the information.
The "Enable Prereleases" switch and also the bug was not known to me.

JohanD78, 2020-06-07:
The bugs i mentioned are unfurtunatly still present in 20200606.

ccrisan, 2020-06-07:
They are probably related to bugs that have been recently fixed in Motion. It's now time to do a major upgrade to all packages (including a merge with upstream Buildroot).

wheelocknet, 2021-04-10:
How?

starbasessd, 2021-04-10:
Go to Settings, Expert Settings, Enable Pre-Releases
Then go to Settings, General, Check for Updates.

wheelocknet, 2021-04-10:
Super fast response. Thanks!

wheelocknet, 2021-04-10:
OK -Update to latest pre-release went smooth until after reboot and now the admin panel shows up very dim and inoperative. Any suggestions before I revert to latest release.

starbasessd, 2021-04-10:
did you clear cache and cookies, and do a switch user to admin?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

wheelocknet, 2021-04-10:
No - just tried with my phone browser and Win 10 Edge and it works beautifully - so it is my Chrome browser problem. Thanks for the fix. I will play with it some more but It seems as this ticket is closed. Great work!

closed 2022-01-19 by starbasessd