motioneye-project/motioneyeos #2416: no-cache" - /etc/init.d/S50date

kristofejro, 2020-06-08:
In release 20200606 there is something wrong with /etc/init.d/S50date
I have observed that two of my motioneyeOS systems are trying to connect to:

```
no-cache"
no-cache".lan
```
In previous stable realeas 20190911 there was no such issue.
I think this is board independent because it is present on RPi Zero W and RPi 2.

ccrisan, 2020-06-12:
Could you please share your log files that indicate these connection attempts?

kristofejro, 2020-06-12:
Here you go:

```
Jun 12 00:00:55 dnsmasq[566]: query[A] no-cache".lan from 192.168.0.128
Jun 12 00:00:55 dnsmasq[566]: forwarded no-cache".lan to 127.0.0.1
Jun 12 00:00:55 dnsmasq[566]: query[AAAA] no-cache".lan from 192.168.0.128
Jun 12 00:00:55 dnsmasq[566]: forwarded no-cache".lan to 127.0.0.1
Jun 12 00:00:55 dnsmasq[566]: query[A] no-cache" from 192.168.0.128
Jun 12 00:00:55 dnsmasq[566]: config no-cache" is NODATA-IPv4
Jun 12 00:00:55 dnsmasq[566]: query[AAAA] no-cache" from 192.168.0.128
Jun 12 00:00:55 dnsmasq[566]: config no-cache" is NODATA-IPv6
Jun 12 00:15:56 dnsmasq[566]: query[A] no-cache".lan from 192.168.0.128
Jun 12 00:15:56 dnsmasq[566]: forwarded no-cache".lan to 127.0.0.1
Jun 12 00:15:56 dnsmasq[566]: query[AAAA] no-cache".lan from 192.168.0.128
Jun 12 00:15:56 dnsmasq[566]: forwarded no-cache".lan to 127.0.0.1
Jun 12 00:15:56 dnsmasq[566]: query[A] no-cache" from 192.168.0.128
Jun 12 00:15:56 dnsmasq[566]: config no-cache" is NODATA-IPv4
Jun 12 00:15:56 dnsmasq[566]: query[AAAA] no-cache" from 192.168.0.128
Jun 12 00:15:56 dnsmasq[566]: config no-cache" is NODATA-IPv6

Jun 12 00:01:41 dnsmasq[566]: query[A] no-cache".lan from 192.168.0.129
Jun 12 00:01:41 dnsmasq[566]: forwarded no-cache".lan to 127.0.0.1
Jun 12 00:01:41 dnsmasq[566]: query[AAAA] no-cache".lan from 192.168.0.129
Jun 12 00:01:41 dnsmasq[566]: forwarded no-cache".lan to 127.0.0.1
Jun 12 00:01:41 dnsmasq[566]: query[A] no-cache" from 192.168.0.129
Jun 12 00:01:41 dnsmasq[566]: config no-cache" is NODATA-IPv4
Jun 12 00:01:41 dnsmasq[566]: query[AAAA] no-cache" from 192.168.0.129
Jun 12 00:01:41 dnsmasq[566]: config no-cache" is NODATA-IPv6
Jun 12 00:16:42 dnsmasq[566]: query[A] no-cache".lan from 192.168.0.129
Jun 12 00:16:42 dnsmasq[566]: forwarded no-cache".lan to 127.0.0.1
Jun 12 00:16:42 dnsmasq[566]: query[AAAA] no-cache".lan from 192.168.0.129
Jun 12 00:16:42 dnsmasq[566]: forwarded no-cache".lan to 127.0.0.1
Jun 12 00:16:42 dnsmasq[566]: query[A] no-cache" from 192.168.0.129
Jun 12 00:16:42 dnsmasq[566]: config no-cache" is NODATA-IPv4
Jun 12 00:16:42 dnsmasq[566]: query[AAAA] no-cache" from 192.168.0.129
Jun 12 00:16:42 dnsmasq[566]: config no-cache" is NODATA-IPv6
```

These are examples for two hosts.

ccrisan, 2020-07-01:
@kristofejro is this a clean setup or an upgrade? I can't seem to be able to reproduce the issue

kristofejro, 2020-07-02:
@ccrisan both were clean install of 20190911 and upgraded to 20200606.

ccrisan, 2020-07-02:
@kristofejro does that happen with a clean install of 20200606?

kristofejro, 2020-07-03:
@ccrisan I have changed date method to SNTP (recommended) and problem is gone.

closed 2020-07-03 by kristofejro