motioneye-project/motioneyeos #2424: unable to open video device

lodgemoorlad, 2020-06-12:
Hi
I have had ME running on my Pi for about 2-3 years with no problem. Not long ago it just stopped working so today I burned another fresh image and fired it up.

I managed to get into the setup screen but when I picked a camera, the default was " local V4L2 " camera so I picked that. However the one I have is a Noir which plugs directly into the Pi board, could this be the problem, should I pick another.
Any help would be very much appreciated. 
Thanks in advance

starbasessd, 2020-06-12:
Try mmal

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lodgemoorlad, 2020-06-12:
Starbasessd, you're a STAR....
Don't understand it though because it did'nt work before. Ah, well it does now and I'm very grateful, thanks, really appreciate it :-)

ferror18, 2020-07-05:
# Thanks in advance to anyone that wants to help me
### I really need to fix this, there is a creepy dude that comes around our house, we believe he stole one of our bikes and I need to set up this cameras to record in case he tries to steal something else.

### Hi im building a pi Zero W camera set up  + pi4 hub server but I im stuck at getting my camera to work just on pi zero.

## WHAT I HAVE DONE 

- Downloaded this image MotionEyeOS(20200606) and burned it into a 16GB SD Card, using balenaEtcher. 

- Later I added my "wpa_supplicant.conf" file to root. 

- Connected my camera (OV5467)

- Connected power, keyboard and monitor,

- Then I plugged this SD Card into my Pi Zero W.

- Booted Successfully

- Went to web ui added new camera.

- Camera appears grey with the following message "UNABLE TO OPEN VIDEO DEVICE"

## ADDITIONAL INFORMATION 

#### I have two ribbon cables I;ve tested both of them and made sure they're plugged in properly I havent gotten around to  testing the camera yet-----------------------------------

#### All of this command have been run from root as I logged in as admin -----------------------------------

- running modprobe bcm2835-v4l2 does not return an error (I believe I dont need to do this but I did and it didnt solved it)

- dmesg | grep v4l           return --->>>       ```[    3.872920] bcm2835_v4l2: module is from the staging directory, the quality is 
    unknown, you have been warned.```

- vcgencmd get_camera          returns --->>>      ```supported=1 detected=0```

- raspistill -v -o test.jpg   returns --->>>>

     ```
     mmal: Cannot read camera info, keeping the defaults for OV5647

     "raspistill" Camera App (commit )

     Camera Name OV5647
     Width 2592, Height 1944, filename test.jpg
     Using camera 0, sensor mode 0

     GPS output Disabled

     Quality 85, Raw no
     Thumbnail enabled Yes, width 64, height 48, quality 35
     Time delay 5000, Timelapse 0
     Link to latest frame enabled  no 
    Full resolution preview No
     Capture method : Single capture

     Preview Yes, Full screen Yes
     Preview window 0,0,1024,768
    Opacity 255
     Sharpness 0, Contrast 0, Brightness 50
     Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
     Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
     Flicker Avoid Mode 'off'
     Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
      Rotation 0, hflip No, vflip No
     ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000 
     mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
     mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
     mmal: Failed to create camera component
     mmal: main: Failed to create camera component
     mmal: Camera is not detected. Please check carefully the camera module is installed correctly

     ```
- lsusb    returns    ---->>>>>>

    ``` Bus 001 Device 001: ID 1d6b:0002
     Bus 001 Device 002: ID 1532:0243
     ```

## LOG FILES TL;DR FULL LOG FILE ATTACHED

- motioneye =>
     ```
     ERROR: mjpg client timed out receiving data for camera 1 on port 8081
     ```
- motion   ====>>>>
     ```
      [1:ml1:Camera1] [ERR] [VID] v4l2_device_capability: Device does not support capturing.
      [1:ml1:Camera1] [ERR] [VID] vid_start: V4L2 device failed to open
      [1:ml1:Camera1] [WRN] [ALL] motion_init: Could not fetch initial image from camera 
      [1:ml1:Camera1] [WRN] [ALL] motion_init: Motion continues using width and height from config file(s)
     ```
- dmesg      =====>>>>>>>
     ~~~
     [    6.236021] udevd[262]: starting version 3.2.7
     [    6.265073] random: udevd: uninitialized urandom read (16 bytes read)
     [    6.267057] random: udevd: uninitialized urandom read (16 bytes read)
     [    6.267360] random: udevd: uninitialized urandom read (16 bytes read)     
     [    6.342329] udevd[263]: starting eudev-3.2.7
     [    7.343812] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
     [    7.394475] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
     [    7.394549] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
     [    7.401400] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
     [    7.401452] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
     [    7.422870] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
     [    7.422923] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
     [    8.219172] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
     ~~~
-  messages  =====>>>>>>
     ~~~
     [    6.236021] udevd[262]: starting version 3.2.7
     [    6.265073] random: udevd: uninitialized urandom read (16 bytes read)
     [    6.267057] random: udevd: uninitialized urandom read (16 bytes read)
     [    6.267360] random: udevd: uninitialized urandom read (16 bytes read)
     [    6.342329] udevd[263]: starting eudev-3.2.7
     [    7.343812] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
     [    7.394475] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
     [    7.394549] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
     [    7.401400] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
     [    7.401452] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
     [    7.422870] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
     [    7.422923] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
     [    8.219172] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
     ~~~

starbasessd, 2020-07-05:
No need to have both of these tickets open. The motioneye group is for if
you are running motioneye, on another OS, the motionEyeOS github is
specifically for motionEyeOS issues.
Pick one and close the other, and we'll try to help you.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

theMrTux, 2021-01-25:
So I'm having the very same issue that ferror18 has. But according to starbasessd, I need to find another forum? Could someone please tell me where to get support for motion and motioneye?

starbasessd, 2021-01-25:
https://github.com/ccrisan/motioneye/issues for motioneye on another OS

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2022-01-19:
No comments after more than 60 days. If you are still having issues, please open a new issue.

closed 2022-01-19 by starbasessd

theventurousco, 2022-08-10:
I'm having the same issue. Running motioneyeos on a rasperry Pi Zero W.
It was working, then one day, poof, grey square and unable to open video device

starbasessd, 2022-08-10:
@theventurousco 
Did you re-image your SDCard with 20200606 or dev20201026?
What kind of camera are you using? CSI ribbon or USB or IP?
If you can access via SSH, what do you get for the command:
> vcgencmd get_throttled

king-obvious-iii, 2022-08-25:
@starbasessd I am having exactly this same issue.  Imaged with 20200606, when I type vcgencm get_throttled I get:
throttled=0x0

when I type in vcgencmd get_camera I get
supported=1 detected=0

thanks in advance

ramoneiggy, 2022-09-24:
Had the same problem. Worked for a while the stopped working after updating to buster.
Found a solution online somewhere where there was a different problem.
I had to change a property inside etc/motioneye/camera-1.conf
videodevice /dev/video10 to video0

video10 was set automatically on first initial setup.
After restart it worked, hope it helps someone.

egzumer, 2022-10-26:
I have exactly the same problem as [king-obvious-iii](https://github.com/king-obvious-iii)

starbasessd, 2022-10-26:
A response to
> vcgencmd get_camera

of 
> supported=1 detected=0

means it's not seeing the camera. Several things to check:
Ribbon connected correctly? The protector side goes away from the board at both ends.
Does the camera work with RaspberryPiOS Buster (Legacy)? Don't use Bullseye, as it doesn't support the Python2 functions needed by motionEye, so is not a valid test.
Is the light on on the camera board? If not, it's not getting enough power from the Pi.

egzumer, 2022-10-26:
Thank you for your answer and sorry for the trouble. When I switched SD cards the camera lost electrical connection, I got fouled because it worked just a couple of minutes ago. I reconnected all connectors and it works now.

starbasessd, 2022-10-26:
@egzumer   Glad it was something simple...

JosephMilo, 2023-03-19:
I'm receiving an "Unable to open video device" error.

I attempted using a Logi C920 and and C270. 

My Install Flow: 

$ sudo apt install motion
$ sudo apt install python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev
$ sudo pip install motioneye
$ sudo mkdir -p /etc/motioneye
$ sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
$ sudo mkdir -p /var/lib/motioneye
$ sudo cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
$ sudo systemctl daemon-reload
$ sudo systemctl enable motioneye
$ sudo systemctl restart motioneye

Interface is working and i'm able to find the cameras in the ME client and /dev/ so it seems to me they're mounting correctly. 

I added to the install thinking there were missing dependencies:
 
$ sudo systemctl stop motioneye
$ sudo apt-get install python2 -y
$ curl https://bootstrap.pypa.io/pip/2.7/get-pip.py --output get-pip.py
$ python2 get-pip.py
$ sudo apt-get install libffi-dev libzbar-dev libzbar0 -y
$ sudo pip2 install motioneye --upgrade
$ sudo systemctl daemon-reload
$ sudo systemctl enable motioneye
$ sudo systemctl start motioneye

I still had the same issue, not sure what's wrong. 

Debug:

$ vcgencmd get_camera
supported=0 detected=0

$ lsusb
Bus 001 Device 059: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 001 Device 002: ID 0000:0000
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ dmesg | grep v4l
<No Output>

$raspistill -v -o test.jpg
mmalipc: mmal_vc_init_fd: could not open vchiq service
mmal: mmal_vc_component_create: failed to initialise mmal ipc for 'vc.camera_inf                              o' (7:EIO)
mmal: mmal_component_create_core: could not create component 'vc.camera_info' (7                              )
mmal: Failed to create camera_info component

"raspistill" Camera App (commit )

Camera Name OV5647
Width 2592, Height 1944, filename test.jpg
Using camera 0, sensor mode 0

GPS output Disabled

Quality 85, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Time delay 5000, Timelapse 0
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmalipc: mmal_vc_init_fd: could not open vchiq service
mmal: mmal_vc_component_create: failed to initialise mmal ipc for 'vc.ril.camera                              ' (7:EIO)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (7)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not enabled in this build. Try running "sudo raspi-config" and ensure that "camera" has been enabled

Any help would be amazing!

starbasessd, 2023-03-19:
A) you are hijacking a CLOSED, wrong issue.
B) You don't state what OS you are running, on what hardware (PC or Pi or Pi-work-alike)
You need to follow the steps in order, for the OS used, over on the correct github. 
This github is for the motionEyeOS which is dedicated to a very light weight install all self contained...
[Here is a link to the correct motionEye github install instructions](https://github.com/motioneye-project/motioneye/wiki/Installation)

JosephMilo, 2023-03-19:
I see camera is not enabled. Enabled it through raspi. 

New Debug:

$ vcgencmd get_camera
supported=1 detected=0

 $ sudo raspistill -v -o test.jpg
mmal: Cannot read camera info, keeping the defaults for OV5647

"raspistill" Camera App (commit )

Camera Name OV5647
Width 2592, Height 1944, filename test.jpg
Using camera 0, sensor mode 0

GPS output Disabled

Quality 85, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Time delay 5000, Timelapse 0
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Flicker Avoid Mode 'off'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not detected. Please check carefully the camera module is installed correctly

starbasessd, 2023-03-19:
The raspi-config enable camera is for CSI cameras (ribbon connector) and is not needed for USB cameras. Please take your issue over to the [motionEye github](https://github.com/motioneye-project/motioneye/wiki/Installation)

JosephMilo, 2023-03-19:
Ah alright. Sorry just learning here. I'll delete out as to not junk up the thread.

TheFax, 2023-09-29:
> Try mmal

Could you tryto explain better please?
I have the same problem, but I cannot what you means.

chwiejak, 2023-11-28:
> Had the same problem. Worked for a while the stopped working after updating to buster. Found a solution online somewhere where there was a different problem. I had to change a property inside [...]

Hi, how to edit metioned file?

genevish, 2024-12-30:
> > Had the same problem. Worked for a while the stopped working after updating to buster. Found a solution online somewhere where there was a different problem. I had to change a property inside etc/motioneye/camera-1.conf videodevice /dev/video10 to video0
> > video10 was set automatically on first initial setup. After restart it worked, hope it helps someone.
> 
> Hi, how to edit file etc/motioneye/camera-1.conf ?

From the terminal:

sudo nano /etc/motioneye/camera-1.conf 

Make the change indicated, then Ctrl-x and "Y" to save.