motioneye-project/motioneyeos #2425: Unable to opne video device - Victure PC650

matteopiccioni, 2020-06-12:
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
Yes

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
Yes

### motionEyeOS Version
I am running motionEyeOS version: 
motionEye Version | 0.42.1
Motion Version | 4.2.2
OS Version | Raspbian 10

### Board Model
I am using the following board/model: Raspberry 4

### Camera
I am using the following type of camera: Network Camera
My camera model is: Victure PC650

### Network Connection
My motionEyeOS unit is connected to the network via: Ethernet

My problem
I am able to add my camera to motionEye, but when config is done, the camera preview show me the message: Unable to open video device
Using the same rtsp url rtsp://ipcamera I am able to see camera using VLC (after insert username e password)
I try more rtsp url for my PC650 using the ones suggested here: https://www.ispyconnect.com/man.aspx?n=Victure#

Thanks
Matteo

starbasessd, 2020-06-12:
How are you entering the URL for the camera? Exact text please.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

matteopiccioni, 2020-06-12:
While Adding Camera I select:

Camera Type: Network Camera
URL: rtsp://192.168.2.204/
Username: myuser
Password: mypass
Camera: H264DVR 1.0RTSP/TCP Camera

Others URL I try to add are:
rtsp://192.168.2.204/live/ch00_0/
rtsp://192.168.2.204/h264/
rtsp://192.168.2.204:554/
rtsp://192.168.2.204:554/live/ch00_0/
rtsp://192.168.2.204:554/h264/
rtsp://192.168.2.204
rtsp://192.168.2.204/live/ch00_0
rtsp://192.168.2.204/h264
rtsp://192.168.2.204:554
rtsp://192.168.2.204:554/live/ch00_0
rtsp://192.168.2.204:554/h264

starbasessd, 2020-06-12:
Did you try  *rtsp://admin:admin@<ip_address>:5544/live/ch00_0   (without
the '<' & '>'... with all your other options....)*

*It's one of those weird cameras. It may not be possible to do it.*

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

matteopiccioni, 2020-06-12:
Unfortunately while adding camera I cant add username and pwd into URL text

So I try to add settings as usual...
Then in my camera-1.conf file
I modify netcam_url values
and remove netcam_userpass values

Unfortunately after restarting motioneye service the camera still show 'Unable to open vide device'
Into Camera Device settings I can see my username and password in clear text ofc

starbasessd, 2020-06-12:
One last test, try as http: instead of rtsp: with the options you tried
above.
Unless someone else has one, I vote for unusable.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

matteopiccioni, 2020-06-12:
I try using
http://admin:mypwd@<ip_address>:544/live/ch00_0
but it doesn't works

Maybe is movie_codec the problem? I dont know...using vlc it works
Thanks

starbasessd, 2020-06-12:
No, a normal setup, using http://ip_address...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

matteopiccioni, 2020-06-12:
I try to:
- add a new network camera using http://192.168.2.204:554/ or http://192.168.2.204
Under Camera it say:  not a supported network camera
- I try to modify the previous camera-1.conf
restoring netcam_userpass and modifying netcam_url with http://
But it doesn't works

starbasessd, 2020-06-12:
Sorry, then, I vote unsupportable, at least until next version.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

matteopiccioni, 2020-06-12:
I just see now that adding camera using UDP protocol works

starbasessd, 2020-06-12:
Interesting. Post your settings to help others.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

matteopiccioni, 2020-06-12:
I add camera using this settings:

Camera Type: Network Camera
URL: rtsp://192.168.2.204/
Username: myuser
Password: mypass
Camera: H264DVR 1.0RTSP/**UDP** Camera

starbasessd, 2020-06-12:
Thanks, good to know!

Don't forget to close the ticket.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

matteopiccioni, 2020-06-12:
The need to be in UDP has some disadvatages?

starbasessd, 2020-06-12:
Up to yeu. If it works, it works for that camera...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

matteopiccioni, 2020-06-12:
Ok, 
thanks for support :)

closed 2020-06-12 by matteopiccioni