motioneye-project/motioneyeos #2427: 20200606 setup trying to write read-only FS; fails, reboots

watou, 2020-06-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

20200606

### Board Model

PINE64 original Kickstarter version; has run many previous images no problem.

### Network Connection

Ethernet

### Log Files

```log
[root@meye-ca3fcea2 log]# cd /tmp
[root@meye-ca3fcea2 tmp]# cat _boot.log
---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Starting eudev: done
 * Starting rngd: failed
 * Preparing AP interfaces: none
 * Configuring wired network: dhcp
 * Setting current date using SNTP: failed
 * Starting ntpd: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: failed
 * Starting nmbd: mkdir failed on directory /var/log/samba/cores: No such file or directory
Failed to create /var/log/samba/cores for user 0 with mode 0700
Unable to setup corepath for nmbd: No such file or directory
[2020/06/13 07:38:59.279609,  0] ../lib/util/debug.c:1063(reopen_logs_internal)
  Unable to open new log file '/var/log/samba/log.nmbd': No such file or directory
[2020/06/13 07:38:59.279800,  0] ../lib/util/util.c:217(directory_create_or_exist)
  mkdir failed on directory /var/lib/samba/private/msg.sock: No such file or directory
failed
 * Starting motioneye: [root@meye-ca3fcea2 tmp]# 
[root@meye-ca3fcea2 tmp]# Connection to XX.XX.XX.XX closed by remote host.
```

ccrisan, 2020-06-13:
1. Is this a fresh install or an upgrade?
2. Do you have the `/var/log` folder? What does `ls /var/log` say?
3. What does `mount | grep /var` say?

watou, 2020-06-13:
1. fresh install with [this image](https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-pine64-20200606.img.xz)
1. There is a `/var/log` directory.  I think the session below should answer your questions but please ask any more if it would help isolate the issue.  Thank you.

```log
[root@meye-ca3fcea2 ~]# ls -al /var/log
total 6
drwxr-xr-x    2 root     root          1024 Jun 13 08:12 ./
drwxr-xr-x    6 root     root          1024 Jun  6 15:27 ../
-rw-r--r--    1 root     root          1380 Jun 13 08:12 boot.log
-rw-r--r--    1 root     root          1257 Jun 13 08:12 messages
[root@meye-ca3fcea2 ~]# cd /var/log
[root@meye-ca3fcea2 log]# cat boot.log
---- booting motionEyeOS 20200606 ----
 * Detecting disk device: /dev/mmcblk0
 * Mounting filesystems: done
 * Loading kernel modules: done
 * Setting hostname: done
 * Starting syslogd: done
 * Starting eudev: done
 * Starting rngd: failed
 * Preparing AP interfaces: none
 * Configuring wired network: dhcp
 * Setting current date using SNTP: failed
 * Starting ntpd: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting smbd: failed
 * Starting nmbd: mkdir failed on directory /var/log/samba/cores: No such file or directory
Failed to create /var/log/samba/cores for user 0 with mode 0700
Unable to setup corepath for nmbd: No such file or directory
[2020/06/13 08:10:04.266929,  0] ../lib/util/debug.c:1063(reopen_logs_internal)
  Unable to open new log file '/var/log/samba/log.nmbd': No such file or directory
[2020/06/13 08:10:04.267132,  0] ../lib/util/util.c:217(directory_create_or_exist)
  mkdir failed on directory /var/lib/samba/private/msg.sock: No such file or directory
failed
 * Starting motioneye: failed
---- shutting down motionEyeOS 20200606 ----
 * Stopping motioneye: failed
 * Stopping smbd: failed
 * Stopping nmbd: failed
 * Stopping proftpd: done
 * Stopping sshd: done
 * Stopping crond: done
 * Stopping ntpd: done
 * Stopping network: done
 * Stopping rngd: done
 * Stopping eudev: done
 * Stopping syslogd: done
[root@meye-ca3fcea2 log]# cat messages
Jun 13 08:10:28 meye-ca3fcea2 daemon.info ntpd[1129]: Listen normally on 6 eth0 [fd7c:13d:3d34:1:10c8:caff:fe3f:cea2]:123
Jun 13 08:10:34 meye-ca3fcea2 cron.err crond[1142]: time disparity of 5495530 minutes detected
Jun 13 08:10:34 meye-ca3fcea2 cron.err crond[1145]: time disparity of 5495530 minutes detected
Jun 13 08:12:10 meye-ca3fcea2 daemon.info : starting pid 1815, tty '': '/etc/init.d/rcK'
Jun 13 08:12:11 meye-ca3fcea2 auth.info sshd[1207]: Received signal 15; terminating.
Jun 13 08:12:11 meye-ca3fcea2 auth.info sshd[1234]: Exiting on signal 15
Jun 13 08:12:11 meye-ca3fcea2 daemon.notice ntpd[1129]: ntpd exiting on signal 15 (Terminated)
Jun 13 08:12:11 meye-ca3fcea2 daemon.info ntpd[1129]: 127.127.1.0 local addr 127.0.0.1 -> <null>
Jun 13 08:12:11 meye-ca3fcea2 daemon.info ntpd[1129]: 162.159.200.123 local addr 10.0.1.25 -> <null>
Jun 13 08:12:11 meye-ca3fcea2 daemon.info ntpd[1129]: 40.112.75.180 local addr 10.0.1.25 -> <null>
Jun 13 08:12:11 meye-ca3fcea2 daemon.info ntpd[1129]: 188.125.64.6 local addr 10.0.1.25 -> <null>
Jun 13 08:12:11 meye-ca3fcea2 daemon.notice proftpd[1220]: localhost - ProFTPD killed (signal 15)
Jun 13 08:12:11 meye-ca3fcea2 daemon.notice proftpd[1220]: localhost - ProFTPD 1.3.6 standalone mode SHUTDOWN
[root@meye-ca3fcea2 log]# mount | grep /var
[root@meye-ca3fcea2 log]# mount | grep \/var
[root@meye-ca3fcea2 log]# mount             
rootfs on / type rootfs (rw)
/dev/root on / type ext4 (ro,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=978104k,nr_inodes=244526,mode=755)
proc on /proc type proc (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p3 on /data type ext4 (rw,noatime,data=ordered)
/dev/mmcblk0p3 on /home/ftp/sdcard type ext4 (rw,noatime,data=ordered)
/dev/mmcblk0p3 on /home/ftp/storage type ext4 (rw,noatime,data=ordered)
[root@meye-ca3fcea2 log]# cat messages
Jun 13 08:10:28 meye-ca3fcea2 daemon.info ntpd[1129]: Listen normally on 6 eth0 [fd7c:13d:3d34:1:10c8:caff:fe3f:cea2]:123
Jun 13 08:10:34 meye-ca3fcea2 cron.err crond[1142]: time disparity of 5495530 minutes detected
Jun 13 08:10:34 meye-ca3fcea2 cron.err crond[1145]: time disparity of 5495530 minutes detected
Jun 13 08:12:10 meye-ca3fcea2 daemon.info : starting pid 1815, tty '': '/etc/init.d/rcK'
Jun 13 08:12:11 meye-ca3fcea2 auth.info sshd[1207]: Received signal 15; terminating.
Jun 13 08:12:11 meye-ca3fcea2 auth.info sshd[1234]: Exiting on signal 15
Jun 13 08:12:11 meye-ca3fcea2 daemon.notice ntpd[1129]: ntpd exiting on signal 15 (Terminated)
Jun 13 08:12:11 meye-ca3fcea2 daemon.info ntpd[1129]: 127.127.1.0 local addr 127.0.0.1 -> <null>
Jun 13 08:12:11 meye-ca3fcea2 daemon.info ntpd[1129]: 162.159.200.123 local addr 10.0.1.25 -> <null>
Jun 13 08:12:11 meye-ca3fcea2 daemon.info ntpd[1129]: 40.112.75.180 local addr 10.0.1.25 -> <null>
Jun 13 08:12:11 meye-ca3fcea2 daemon.info ntpd[1129]: 188.125.64.6 local addr 10.0.1.25 -> <null>
Jun 13 08:12:11 meye-ca3fcea2 daemon.notice proftpd[1220]: localhost - ProFTPD killed (signal 15)
Jun 13 08:12:11 meye-ca3fcea2 daemon.notice proftpd[1220]: localhost - ProFTPD 1.3.6 standalone mode SHUTDOWN
[root@meye-ca3fcea2 log]# mount | grep /var
[root@meye-ca3fcea2 log]# cat boot.logConnection to 10.0.1.25 closed by remote host.
```

danilo-jlle, 2020-06-20:
I confirm this issue.
Mine is a little worse.
https://photos.app.goo.gl/EGttCQMLShjqMnT58
[image]

watou, 2020-06-30:
@danilo-jlle Thank you very much for trying older images and saving me from repeating the same steps you took.  Hopefully PINE64 support has sufficient priority to be supported again.

mgg11, 2021-04-07:
Confirming I'm seeing the same issue on my Pine64+.
Is there any solution to this, other than running an older version?

starbasessd, 2021-04-07:
The odroid c boards and the Pine64 were not built for dev20201026. May I suggest doing a 'build your own' version from the instructions here:
https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source
I've done so, and it works. If you would like me to test the docker build version, I can do so.

mgg11, 2021-04-07:
@starbasessd thank you for the tip!
I just tried that, following the instruction, but pine64 seems to be an "unknown board". It's not part of the repository for some reason.

starbasessd, 2021-04-07:
Unfortunately, I don't have one of the older version's configs/pine*_defconfig and configs/pine*_initramfs_defconfig files, otherwise I'd pop it in and see of it builds against the current stuff, sorry. THey're no longer in the SourceCode configs folder, either. Strange.

watou, 2021-04-10:
@starbasessd I see the old `configs/pine64_defconfig` and `configs/pine64_initramfs_defconfig` files in [this tag](https://github.com/ccrisan/motioneyeos/tree/b5ceadc43db53d3217524522bb90b54672107b83/configs) (removed with [this commit](https://github.com/ccrisan/motioneyeos/commit/fab0316f7cc97912769eae342cb878ca0a0b73b8#diff-175ab01246347ee3cb3ad67cd8f359e5bdd765f7751c692026b7e782a3fa53ef)), but I don't know if those versions are compatible with current.  I would love to get my Kickstarter PINE64 running motionEyeOS again if there is any path forward.

starbasessd, 2021-04-10:
If those older files don't work, then the only thing I can recommend is to
run a Pine64 Lite OS and install motion/motionEye. Sorry.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2022-01-19 by starbasessd