motioneye-project/motioneyeos #2428: Cannot create an access point with a Pi Zero W

vonvogel007, 2020-06-13:
I can't manage to create an AP on the Pi Zero W!
- motioneyeos-raspberrypi-20200606.img.xz
- I placed 4 configuration files in the boot partition (practically copy-pasting from the wiki): os.conf (to set networkless to true), dnsmasq.conf, static_ip.conf and hostapd.conf.

dnsmasq.conf
`interface=wlan0
dhcp-range=192.168.27.50,192.168.27.150,24h
dhcp-leasefile=/data/dnsmasq.leases`

hostapd.conf
`ssid=Test007
wpa_passphrase=12345678
wpa=2
wpa_key_mgmt=WPA-PSK
interface=wlan0
channel=1
driver=nl80211
hw_mode=g
ieee80211n=1
ieee80211ac=1`

os.conf
`OS_DEBUG="false"
OS_PRERELEASES="false"
OS_TTY_LOGIN="tty1"
OS_ETH="eth0"
OS_WLAN="wlan0"
OS_PPP="ppp0"
OS_NETWORKLESS="true"
OS_COUNTRY="GB"
OS_FIRMWARE_METHOD="github"
OS_FIRMWARE_REPO="ccrisan/motioneyeos"
OS_FIRMWARE_USERNAME=""
OS_FIRMWARE_PASSWORD=""
`

static_ip.conf
`#STATIC_IP="192.168.1.101/24"
#STATIC_GW="192.168.1.1"
#STATIC_DNS="8.8.8.8"
`
The zero boots, but I can't see any WLAN "Test007" with my notebook or cell phone.
With a different image, it works with a Pi 3 without problems.

greetings

Wolfgang

starbasessd, 2020-06-13:
A little better set of instructions and sample files.

[ConfigFiles.zip](https://github.com/ccrisan/motioneyeos/files/4774879/ConfigFiles.zip)

starbasessd, 2020-06-13:
BTW, can't edit the files with Notepad, need to use NotePad++ or similar Linux Friendly Windows app. MacOS and Linux don't have issues.

vonvogel007, 2020-06-13:
Unfortunately, no WiFi "picam" is found in the Notbook / Phone!
- I flashed the IMG onto the SD card
- copied the 4 new files into the boot directory
- Waited until the green LED stops blinking.

What I have to say, I did my first tests with the Zero using the WLAN that the Zero dialed into. That worked immediately, but he doesn't create an AP!

Maybe there is a problem with the current IMG?

starbasessd, 2020-06-13:
Confirm it with using 20200203 at:
https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi-20200203.img.xz

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

vonvogel007, 2020-06-13:
Success! 
It works with the motioneyeos-raspberrypi-20200203.img.xz! 
Many thanks for your help!

greetings

starbasessd, 2020-06-13:
@ccrisan another thing to check, sir.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2020-06-13:
Will take a look at it.

unstatusthequo, 2020-06-27:
I am having the same issue with `motioneyeos-raspberrypi3-20200606.img`

I've reflashed several times, used nano in OSX to setup the 4 .conf files referenced above, and on first boot it sets up partition, but panics when setting up keys, but continues. This isn't reflected in the boot log. It then reboots itself, gets the static IP from the conf file, then gets to login. I can login via terminal and do whatever, but no SSID appears from the device as an Access Point. I first thought maybe it was hidden, but manually trying to login to that and it's not found.

starbasessd, 2020-06-27:
Try it with 20200203 & these files.
As previously reported there may be issues with 20200606

[ConfigFiles.zip](https://github.com/ccrisan/motioneyeos/files/4839989/ConfigFiles.zip)

mhaserodt, 2020-07-01:
I just wanted to add that I'm experiencing the same thing.  I'm unable to setup as a WiFi AP under 20200606, but 20200203 and prior versions work fine.  I've got the same issue where the network isn't broadcasting and I can't add it manually either.  I've tried it with the ConfigFiles.zip as provided, and my own files that I've been using for a couple of years (basically the same thing, only I turn off the SSID broadcast in hostapd.conf, I use a slightly different IP address in dnsmasq.conf, and I've never used static_ip.conf.)

Thanks for working on this.  Being able to SSH while it's setup as an AP would make things a lot easier for me!

Schillers, 2020-11-09:
Just to add, I am experiencing the same as above.

Many thanks!

starbasessd, 2020-11-09:
motionEyeOS dev20201026 clean install on 8GB sdCard PiuZeroW (known good working)
First Boot: hostapd panics during SSH Key Gen, finishes, reboots. Upon reboot, boot screen loads normally, but never starts broadcasting ssid.
Attempted to recover log files, nothing found in /var/log 
Same installation, with 20200203, works fine.
I notice that AP0 is not mentioned in 20200203 boot screen, but is mentioned in 20200606 & dev20202026
This is a @CCRISAN issue and he is being notified.
In the mean time, if you use 20200203 from https://github.com/ccrisan/motioneyeos/releases
and the attached sample files, it does work.
[ConfigFiles.zip](https://github.com/ccrisan/motioneyeos/files/5512222/ConfigFiles.zip)

starbasessd, 2020-11-09:
OK, the new feature ap0 wasn't documented (now on my list of things to do @ccrisan )
I have modded the config files in the attached ConfigFiles(20200606-or-newer).zip to use the new feature.
[ConfigFiles(20200606-or-newer).zip](https://github.com/ccrisan/motioneyeos/files/5512392/ConfigFiles.20200606-or-newer.zip)
Continue to use the original ConfigFiles.zip for anything earlier than 20200606
[ConfigFiles.zip](https://github.com/ccrisan/motioneyeos/files/5512401/ConfigFiles.zip)
Updated Wiki docs coming soon.

starbasessd, 2020-11-10:
Wiki [Wifi Access Point](https://github.com/ccrisan/motioneyeos/wiki/WiFi-Access-Point) instructions updated, and example files attached to my previous post.

closed 2020-11-10 by starbasessd