motioneye-project/motioneyeos #2429: Feature Request - ISO to work in virtual Machine

corincorvus, 2020-06-16:
Hi,

i use your OS on some Pis. At the Moment my Pi 4 is the Central for all Cameras. Sometimes he overheat and crash with heatsink, cause the load is to high. May you can create a bootable ISO to work on a virtual Machine? So some Users can use it to make his own "Central Hub" to connect all Cams. 

Best Regards

starbasessd, 2020-06-16:
You can run notion & notionEye in a Linux VirtualMachine. I run a
VirtualBox Server that has as one of many, an Ubuntu 20.04 set up running
motion & motionEye, Works well.
You could try running motionEyeOS in a QEMU ARM-HF environment, but there
are many easier ways to get it done.
What is your environment that you are overheating a Pi4? I have 2 now (a
Pi4-4GB and a Pi4-8GB), so could help test to see where it is having
issues. My PI4-4GB hasn't shown any signs
of overheating and all I run on it (as most of my 3B & 3B+s) is a 5mm
heatsink and some mini-fans in aluminum cases.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

corincorvus, 2020-06-17:
Ok i will try ist today.

my pi 4 (4gb) manage at this moment 19 cam streams. Every 15 minutes he goes to 87 °C and the webgui stops, the Shell stucks. After some seconds the pi run fine again. I use the Argon One Case for cooling with active cooling fan.

starbasessd, 2020-06-17:
Any heat sinks? GPU and memory heat sinks might help. But with that many
cameras you might want to try the VM solution...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

corincorvus, 2020-06-17:
VM work :) 
yes i use the "argon one" case with active fan and headsinks with coolpads for all components.
so i close this ticket now. Thank you for fast help.

closed 2020-06-17 by corincorvus