motioneye-project/motioneyeos #2430: Error building from source (libgpg-error-1.33)

JoseAlcerreca, 2020-06-17:
Following the steps in 

https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source

With `./build.sh raspberrypi4 xconfig`

I enabled `mosquitto` and built with `./build.sh raspberrypi4`

(Edit:I started from scratch and it fails even without extra modules after `./build.sh raspberrypi4 clean-target`. Also tried cleaning the repo.)

Got:

```
/path/motioneyeos/motioneyeos/output/raspberrypi4/host/bin/arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I..  -DPKGDATADIR=\"/usr/share/libgpg-error\" -DLOCALEDIR=\"/usr/share/locale\"  -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64  -O2   -Wall -Wpointer-arith -Wno-psabi -fvisibility=hidden -c -o gpg_error-gpg-error.o `test -f 'gpg-error.c' || echo './'`gpg-error.c
strerror-sym.c: In function ‘gpg_strerror_sym’:
strerror-sym.c:45:13: warning: implicit declaration of function ‘errnos_msgidxof’; did you mean ‘msgidxof’? [-Wimplicit-function-declaration]
       idx = errnos_msgidxof (code);
             ^~~~~~~~~~~~~~~
             msgidxof
strerror-sym.c:47:9: error: ‘errnos_msgstr’ undeclared (first use in this function)
  return errnos_msgstr + errnos_msgidx[idx];
         ^~~~~~~~~~~~~
strerror-sym.c:47:9: note: each undeclared identifier is reported only once for each function it appears in
strerror-sym.c:47:25: error: ‘errnos_msgidx’ undeclared (first use in this function)
  return errnos_msgstr + errnos_msgidx[idx];
                         ^~~~~~~~~~~~~
make[4]: *** [Makefile:1120: gpg_error-strerror-sym.o] Error 1
```

Branch: `dev`

JoseAlcerreca, 2020-06-17:
Apparently there's an issue with gawk 5 and the `libgpg-error` package. https://dev.gnupg.org/T4459

jb035, 2021-05-07:
I reference to err-sources.h and change 
from 
idx = errnos_msgidxof (code);
return errnos_msgstr + errnos_msgidx[idx];
to
idx = msgidxof (code);
return msgstr[idx];