motioneye-project/motioneyeos #2431: Embed URL asks for credentials even with Authentication set to Disabled

mandeponium, 2020-06-17:
Maybe I'm not understanding it correctly. However the Embed URL does work when User has a blank password. But with a password the Embed URL asks for credentials.

starbasessd, 2020-06-17:
What are you running? motion/motionEye on another OS? motionEyeOS?
What hardware?
Which version? (Don't say 'latest', be specific 20190911, 20200203,
20200606?)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mandeponium, 2020-06-17:
MotionEyeOS. Version 20200606

starbasessd, 2020-06-17:
And the other info?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mandeponium, 2020-06-17:
Ras pi 3b