motioneye-project/motioneyeos #2432: Images every second with maximum resolution

vonvogel007, 2020-06-17:
Hi!
It's about the Pi Zero W with the PiCam.
Streams are maximum with a resolution of 1920 x 1080, how is it possible to get photos in the resolution of 2592 x 1944 every second instead of the video stream?

Regards

Wolfgang

vonvogel007, 2020-06-18:
Is it not possible to get pictures in full resolution???

Regards

starbasessd, 2020-06-18:
It may be a limitation of 'motion', the back-end, or motionEye, the front
end. You could try running on Rasdpbian OS with the motion/motionEye
install instructions here, and modify the codecs and encoder/decoders to
allow.
@jasaw @ ccrisan should be able to answer better.
Instructions:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd

jasaw, 2020-10-27:
Sorry I missed the question that was addressed to me.

Technically, it's possible to get multiple outputs when using a PiCam, and that's how Fast Network Cam RTSP mode works as well. At the moment, you would need to write your own software to do it.

Here's what your software needs to do:
Set up 2 pipelines in the GPU via the MMAL interface. This is how your pipelines should look like.
```
                                                    +--> ISP (for dropping resolution to 1920 x 1080) --> H264 encoder --> video output
camera output (2592 x 1944 resolution) --> splitter +
                                                    +--> JPEG encoder --> still image output
```