motioneye-project/motioneyeos #2433: File storage mass deletion capability

mihirpatel25, 2020-06-18:
hi,
i am an end user and i have experiecning slower raspberry pie when the storage is above 50% on my sd card.when i try to delete the files, i have to it manually one by one.the GUI does not have the capability of mass deletion or sorted deletion.it would be nice if the file can be deleted in mass by sorting weekly, bi-weekly or monthly basis or by selecting individual file and delete in once.
if developers can add this features would be more than happy and gratefull.
Thanks.

starbasessd, 2020-06-18:
In the meantime, I use ssh, remote into the Pi and I can do the sorting,
filtering, and such there. It would be nice to have that capability in the
GUI, I agree.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mihirpatel25, 2020-06-18:
yes, and as i said,I am an end user.i am teaching my partner to use GUI which is much simpler to use who does not have any prior experience using command prompt.that is why it is needed.

Dewey3, 2020-06-25:
It's may not be considered a "mass" deletion, but you should be able to do "bulk" deletions by day by pressing "Delete All" for the day that is highlighted.

kathgironpe, 2022-05-26:
I agree about deleting videos except for selected days. It will help for a lot.

So my AI-powered cameras are obviously better than anything DIY I came up with. It captured videos of people trying to enter my property. Meanwhile, motioneyeOS has too much captured and I find just scanning through videos challenging.

Currently, there are many ways to mass delete videos but mostly through scripts. 

I have been using the `docker` version for years.

My current solution is too lazy through Bash. Just delete all the files. 
Now I am working on an experiment with AI/ML so it will not delete videos I want to see.

starbasessd, 2022-05-26:
I now have Settings, Movies, Preserve Movies Set to Custom, 3 days (ditto for stills) and it is working well. Manually, I use either SSH or Windows SMB service and filter by date/type/size. Since all 6 of my cameras save to the same NAS themselves, I use the same filename pattern as my cameras, and the Preserve Movies function deletes all files, including the ones the cameras store on the NAS, over 3 days old, every midnight.