motioneye-project/motioneyeos #2434: Unable to open video device Raspberry 3B + Kamera Module V2

grandeivid, 2020-06-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 

motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

### Board Model

I am using the following board/model: Raspberry PI 3B

### Camera

My camera model is: Raspberry Camera V2

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue: none

### Log Files

I consider the following log files relevant to this issue:
 
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4798698/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4798699/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4798700/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4798701/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/4798702/motion.log)

Hi,

I am David. Totally new to the Raspberry world.

I just bought a Raspberry Pi and camera module. I installed MotionEyeOs in the SD Card with no problem. I can power up the board and I can access to the MotionEye interface (using the IP address on Firefox) but I got the message "Unable to open video device". So, there is no image preview.

I looked a lot of threats but I couldn't find a solution. I checked the cables, the connections, rebooted the Raspebrry but no luck.

Can someone please give me a hand with this?

Thank you in advance,

David

starbasessd, 2020-06-18:
Can we try loading 20190911 from
https://github.com/ccrisan/motioneyeos/releases/download/20190911/motioneyeos-raspberrypi3-20190911.img.xz
, please?
Let me know that works or not.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-06-18:
Also, try selecting MMAL camera if using the on-board CSI connector.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

grandeivid, 2020-06-18:
Thank you both.
I am downloading the 20190911 version to give it a try.
Beofre doing that, how can I select the MMAL option? Is it through the MotionEye web interface?

Thank you very much,

David

starbasessd, 2020-06-18:
Yes, delete the current camera, and add a new camera using MMAL instead of
V4L2.
Upper left corner, log in as Admin
Upper Left corner, select the 3 horizontal lines
Upper Left corner, select the trash can
Upper Left Corner select Add Camera

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

grandeivid, 2020-06-18:
I found it, thank you. Althought when I select MMAL is says (no cameras).
If I select Local V4L2, I have the 3 options to select, all of them are the same: "bcm2835-codec-decodec" and all them throw the green screen :)
So, I am going to try with the version of EyeOS you sent me.

I wil let you know if it works.

Thanks a lot for your time

David

grandeivid, 2020-06-18:
Hi again,

Now I have this version:

motionEye Version | 0.41
Motion Version | 4.2.2
OS Version | motionEyeOS 20190911

But, with the same result. I can't not install the camera using the MMAL option. I can only install it selecting Local V4L2, and I got the same grey screen,

Any other ideas?

Thank you

David

starbasessd, 2020-06-18:
Checking.
How is your camera connected? Ribbon cable to CSI port or other?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-06-18:
Which camera are you using, and how is it connected?
Logs do show a V4L2 camera when booting.
My Raspberry Pi Camera V1.3 is auto-detected and shows as a MMAL Camera
Is the 2.0 version the one that just came out? It's possible it's not
supported yet.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

grandeivid, 2020-06-18:
Yes, I just bought the board and the camera and I used the cable that was attached to the camera. I send you 2 pictures, so you can see how it is connected (althought I don't know if I attatched the pictures correctly). But yeah, I used the ribbon cable with the blue end and it is connected to the CSI Port.

[image]
[image]

grandeivid, 2020-06-18:
Sorry I didn't asnwer all your questions.
Yes, I bought the Camera Module V2. And I don't know when it was released. I couldn't find an official list of supported cameras.
If it is not compatible, my goal is to have a video security system. Do you know if there is something similar to the MotionEyeOs for the Raspberry?

starbasessd, 2020-06-18:
Hmmm.
Excellent pictures, BTW, thanks.
Short of loading Raspbian/RaspberyPiOS (
https://downloads.raspberrypi.org/raspios_armhf_latest ) and testing if the
camera works there, it may be that motionEyeOS just hasn't quite caught up
to the latest kernel and drivers for it yet. If it does work there, an
option would be to install RPiOS, then load motion & motionEye using the
instructions here:
https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian
Yes, there are many versions of motionEyeOS for the Raspberry Pi depending
on which PI you have. It is also compatible with the previous versions of
Raspberry Pi Org's camera hardware, and most other network and USB cameras
(but, granted, not all).
I haven't been lucky enough to acquire the new camera module, yet, but it's
on my wish list.
Let me know if you want/need further assistance

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

grandeivid, 2020-06-18:
Wow, and I bought the Raspberry thinking it would be easy to set up the system... :)

All right, I will test the camera first and if it work, I will try it with RPIOS + load motion + motionEye.

Thank you very much and I  hope I will be able to get it running without having to bother you.

Have a nice day !

David

starbasessd, 2020-06-18:
No problem.
BTW, for cameras in RPIOS, don't forget to turn on the CSI interface:
Terminal/Command line
sudo raspi-config
cursor down to Interfacing Options
P1 Camera
Select Enable
(commonly missed option)
It's not bad once you get over this noobie hump...
You will enjoy it!

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

grandeivid, 2020-06-18:
All right, thanks for the tip ! All these sound like chinese to me, but I'll get it running...

Thanks again

grandeivid, 2020-06-18:
Hi again,

I think the camera came with a malfunction. I installed the RPIOS, then, I activated the camera and rebooted the Raspberry. When I try to take picture with the raspistill command, I got the error:

Cannot read camera info, keeping the defaults for 0V5647
mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEN)
Failed to create camera component
Camera is not detected. Please check carefully the camera module is installed correctly

I unpluged and plugged the camera like 3 times, and I am sure it is well connected.
So, I am going to order a new one and see what happens...

Thanks for your time

David

grandeivid, 2020-06-18:
Yes, I already tried that.
Never mind, I will order a new one and takeit from there.
Thanks a lot

Closing the thread...

closed 2020-06-18 by grandeivid