motioneye-project/motioneyeos #2435: Trust certificate for webhook or ignore.

CodedBeard, 2020-06-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 

motionEye Version | 0.42.1
Motion Version | 4.2.2+gitUNKNOWN
OS Version | motionEyeOS 20200606

### Board Model

I am using the following board/model: Raspberry Pi 4B

### Camera

I am using the following type of camera: V4L2

My camera model is: Microsoft Lifecam 6000

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Question
Firstly, thanks for this awesome software. 
I have a webhook configured for motion notification, but the api uses a certificate that motioneyeos does not trust. I managed to get the pem file onto the motioneyeos, however, I can't find any documentation on how to force an update to use this ca pem. I thought it might be the same as raspbian, but the required commands seem to be missing. I've tried searching for motioneyeos specific guide but couldn't find anything. Is adding additional CA's possible, or failing that, simply telling the webhook call to ignore the certificate verification for this url?

starbasessd, 2020-06-18:
Is the webhook using a http or https url?
if https, does it work only calling with http?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

CodedBeard, 2020-06-18:
The webhook is https only, there is no http endpoint. I can see in the logs that it is trying to call the webhook, but failing due to untrusted certificate.
```
2020-06-18 16:04:32: [webhook]    ERROR: failed to call webhook: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)>
```

CodedBeard, 2020-06-19:
Ok, thanks for confirming I wasn't just missing something obvious. For the time being, I've deployed a seperate api that allows http calls, to call the https endpoint on the motioneyeos devices behalf. I'll look at a custom BuildRoot another day when I have more time.

closed 2020-06-19 by CodedBeard