motioneye-project/motioneyeos #2436: File rotation on SD card ?

spinko101, 2020-06-18:
Hi, I have MotionEyeOS running fine on a RaspiPi 3 and 4. Both have 32Gb SD cards to which the movie files are written.

My Q: does MotionEyeOS rotate these files once the storage is full ? I only see an option to "Preserve the files" for a certain period.

Thanks

starbasessd, 2020-06-18:
No, not based on size, just time.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

spinko101, 2020-06-18:
Thanks for your reply. 
May I ask for this feature in a future release ? Size based file management would make more sense imho.

starbasessd, 2020-06-18:
Leave it here for @ccrisan.

Size based on how full the system is, with a trigger min/max (don't start
deleting until XX percentage is left, delete all but the latest YY percent?

Are you trying to make sense? <LOL> :)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

spinko101, 2020-06-19:
yep, something like FIFO.