motioneye-project/motioneyeos #2438: Stream limit?

LeroyMemphis, 2020-06-21:
Hello, maybe someone can help me.

I have got motioneye running on a RPI4.

My DVR supports up to 8 camera streams and all of them are running without problems on VLC,
but when i try to include camera 5-8 in motioneye, i only see the stream of camera 1-4 in camera 5-8.

Thanks for your help.

starbasessd, 2020-06-21:
Usually means corrupted Camera-x.conf files.
Quickest fix is to re-image.
If you want to try to fix it, and know how to ssh into the Pi, or have
console access, you can try to edit or delete the camera-x.conf files.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd