motioneye-project/motioneyeos #2439: Boot motioneyeOS from USB/SSD on Raspberry Pi 4B only

PsyEng, 2020-06-21:
### motionEyeOS Version
I am running motionEyeOS version: 
20200606

### Board Model
I am using the following board/model: 
Raspberry Pi 4B 4GB

### Peripherals
I am using the following peripherals that I consider relevant to this issue:
External SSD with USB-SATA-adapter (GPT formatted)
 

## Here´s a small guide to install and run motionEyeOS on the Raspberry Pi 4B from an external SSD or other mass-storage.

1.) Flash a SD-Card with Raspbian OS and do a full upgrade (apt update && apt full-upgrade) and update your bootloader, following this guide [BETA setup instructions](https://www.raspberryPi.org/documentation/hardware/raspberryPi/bcm2711_bootloader_config.md).

2.) After you´ve verified, that your bootloader_config is correct, flash a SD-Card with the latest motionEyeOS (mine is 20200606) and start the Pi, so it can do the initial setup and build all partitions.

3.) Connect your external device and run "fdisk /dev/sda", press "o" to create an empty DOS partition table, "n" to create 3 partitions. 1. partition starts at "2048" and ends at "80078" (its my config, so be sure its enough space for you), 2. partition starts at "204800" and ends at "655359", third partition starts at "2097152" and ends with the last sector of your device. Now quit with pressing "w" and verify the correct partitions on sda with "fdisk -l /dev/sda" and reboot the Pi. (the start and end are the standard sectors of motionEyeOS, except partition 1 which needs more space for the updated files on boot partition).

4.) Now run "dd of=/mmcblk0p1 if=/dev/sda1 bs=1M", "dd of=/mmcblk0p2 if=/dev/sda2 bs=1M", "dd of=/mmcblk0p3 if=/dev/sda3 bs=1M" and wait till it´s finished.

5.) Now you need another SD-Card with Rasbpian OS or a Linux machine to expand the filesystem of "/dev/sda1", which contains a FAT16 partition. Install with "apt install fatresize" the utility to expand the filesystem. Run the command "fatresize -s 40M /dev/sda1" to resize the partition to the full size, which we created in step 3. (It´s necessary, even if fdisk shows the full size, "df -h" will show less space, after mounting).

6.) Now you can switch back to motionEyeOS and mount all partitions with "mount /dev/sda1 /data/media/sda1", "mount /dev/sda2 /data/media/sda2", "mount /dev/sda3 /data/media/sda3" and verify, all your files are present with "ls -l /data/media/sda*". 

7.) Now navigate to "/data/media/sda2/etc" and run this commands "sed 's/mmcblk0p/sda/ fstab.disk", "sed 's/mmcblk0p/sda/ mtab". Now navigate to "/data/media/sda1" and run this command "sed 's/mmcblk0p/sda/ cmdline.txt". This will replace all mounting points from your SD-Card to your external device.

8.) Now follow this guide [Create a bootable USB drive](https://www.raspberryPi.org/documentation/hardware/raspberryPi/bcm2711_bootloader_config.md) and copy all the needed files to "/data/media/sda1".

9.) Now you can shutdown your Pi, eject the SD-Card and boot from your external device.
 

#### There´s a chance, you´ve to use a GPT-parted device or run in issues with your USB-SATA-adapter. If this is the case, I will add the steps, to get around these problems.

starbasessd, 2020-06-21:
Nice. Will test and confirm as soon as I can.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PsyEng, 2020-06-21:
One thing I will mention, which I forgot in the main post.

I think, but I'm not sure, that system updates, could break something, because some files in "/data/etc" contains references to "mmcblk" and "updates", but my goal was to get the system running. Also there could be references in other binary files, which I haven't controlled.

Maybe I can search in the next days trough the sourcecode and find related files, which has to be modified, so the system is not static bonded to SD-Cards, which is definitely a misconfiguration. It shouldn't be to hard, to set dynamic links for filesystems, other distros does it so and it works just fine. Also there're some files, which contains check for sda, mmcblk, etc.

PsyEng, 2020-06-24:
@starbasessd Did you try it and could confirm it or got some trouble or something else? If yes, could you please tell me, in which problems did you run and could you resolve them by yourself?

starbasessd, 2020-06-24:
On my list for today, sorry. RealLife interfered the past couple of days.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-06-24:
First issue found:
To make 20200606 bootable with any of the most recent bootloader firmwares
you need to replace fixup4.dat and start4.elf with ones dated 5/21/2020 or
later. This applies especially to the 4B-8GB.
I will attach them in a bit.
More to follow.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-06-24:
Replacing the files start4.elf and fixup4.dat allows normal booting,
however on the 4B-8GB doesn't allow discovery or mounting of USB Drives.
Nothing shows in dmesg or other logs when mounting a USB3 thumb drive, or a
USB3 1TB external drive, in either USB3 or USB2 ports.
Interesting. Looking into it further. My 4B-4GB is currently in a
production use, will not be able to test there for a bit (couple of days,
maybe)
More soon.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-06-24:
BTW, I am using
pi@raspberrypi:~ $ sudo rpi-eeprom-update
BCM2711 detected
BOOTLOADER: up-to-date
CURRENT: Mon 15 Jun 13:36:19 UTC 2020 (1592228179)
 LATEST: Thu 16 Apr 17:11:26 UTC 2020 (1587057086)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad
which allows cleanest USB and Network boots
(/lib/firmware/raspberrypi/bootloader/stable for my 4B-8G.
I confirmed that dmesg shows drives being installed in Buster-Lite with no
issues, so not hardware related.
Looking further.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-06-24:
Giving up on Pi4B-8G for now. Cannot get any version or mods of motionEyeOS
to boot. Works fine with several other OSs (might have to copy the
start4.elf & fixup4.dat files, but no other issues. Unable to test RPi4B-4G
at this time.
Might see if I can run a docker build and get a newer version running.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PsyEng, 2020-06-26:
> […](#)
> -- Thanks Kevin Shumaker Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel> N38° 19' 56.52" W85° 45' 8.56" Semper Gumby “Don't tell people how to do things. Tell them [...]

Sry, didn't get notifications for your messages and was quiet busy with work. I've an idea, maybe we can solve this. Do you've telegram or something like this, so we could communicate a little bit faster?

starbasessd, 2020-06-26:
No, but my direct email is in my WIX site, in my signature.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd

lukasig, 2020-12-28:
I am much interested into booting motioneyeOS from usb only for faster response purposes especially, and since I already have an SSD connected to a USB3 port via a sata adapter. But I am struggling to get it working. On another pi4 i am using nextcloudpi, which has a great documentation on how to achieve this on [this page](https://docs.nextcloudpi.com/en/install-ncp-on-pi4-to-external-usb-or-ssd/).

Since I am somewhat a newbie in understanding how this all works, can someone maybe take a look at the page linked above and see if anything from that process can be applied to motioneyeOS?

PsyEng, 2020-12-28:
@lukasig Did you tried to flash it like any other image?
I imagine, the last time I used MotioneyeOS, it was only this...

@starbasessd Sry, I didn't found your mail and got lost on this project, because I'm now using RaspberrypiOS with motion and motioneye for different reasons.
If you didn't get MotioneyeOS running on your 8GB model, I will try to get hand on one and find a way, it shouldn't be to hard, and will contact you directly by mail. I've saved your mail to my personal account, so this time it shouldn't get lost, sry for that.

lukasig, 2020-12-29:
So far I have tried flashing both the sd card and ssd with the latest motioneyeOS using etcher, and I have tried to use the SD card copier through Pi OS desktop.
No luck so far...

lukasig, 2020-12-29:
I am currently trying to see if I can manage your method, but the first issue i ran into came up after running `dd of=/mmcblk0p1 if=/dev/sda1 bs=1M` from step 4. I get `dd: can't open '/mmcblk0p1': Read-only file system` . sudo does not work...

starbasessd, 2020-12-29:
Sudo isn't installed, you are already admin/root when you log in. (nor is
apt, or any other method of adding software. If you need to add
drivers/modules you need to copy the folders from a 'normal' raspbian to
the same location in motionEyeOS, like the ZD1211 wifi drivers, while in
debug mode and enable them. Python scripts can usually be added to the
/data folder tree.
The primary file system is Read Only. This can be modded by switching to
debug mode or remounting rw mode, but defeats a lot of the purpose of
motionEyeOS (protection of the OS from damage due to power failures and
bad reboots)
There are many (too many?) hard coded references to mmcblk0p within the
system.
If you need USB boot, it's currently easier to grab a 'Lite' version of an
OS then install the motion/motionEye on it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PsyEng, 2020-12-29:
> 
> There are many (too many?) hard coded references to mmcblk0p within the system. 

Yes, thats one of the major problems (in my opinion) from MotionEyeOS, but it worked totaly fine on my side.

>
> If you need USB boot, it's currently easier to grab a 'Lite' version of an OS then install the motion/motionEye on it.

Yeah maybe, but this will result in other problems, which I now have.
1. No 1:1 build of ffmpeg and dependicies, which means:
2. No OMX support so 30fps 1080p is unusable (1080p runs about 1-6fps)
2. Green picture in videofiles
3. etc..
4. The configuration files from MotionEyeOS are not 100% compatible to standalone MotionEye and Motion

@lukasig I will try it later, because I´ve to setup my system new, so I´ve some place to test this for you.

lukasig, 2020-12-29:
@PsyEng 
just a quick note on getting green picture video files. I had the same issue but it seems to be solved with dev20201026 version.

PsyEng, 2020-12-29:
> 
> 
> @PsyEng
> just a quick note on getting green picture video files. I had the same issue but it seems to be solved with dev20201026 version.

Yeah, but I´m not using MotionEyeOS anymore, I switched to RaspberryPiOS lite back, cause the Pi doing more then motioneye, so this branch wouldn´t help me at all.

lukasig, 2020-12-29:
I suppose I don't mind using the SD card strictly for booting, but everything else to be handled by the SSD. Would that make the setup process any easier?

PsyEng, 2020-12-30:
@lukasig Okay, I´ve to apologize myself, but got absolutly crazy, cause I couldn´t solve the problem with my pink picture and found over 3 or 4 corners, the solution. It´s discussed in some closed issues, but nowhere documented...
I´ts a pain, if they know, there´re problems on different versions of the different "services" (motionEyeOS, motion and motioneye), which all push the people around the other projects. because their "error" is fixed, but they occured another error on the other project...

Your problem wouldn´t be "easier" if you use the SD-card for boot, it makes it even more worse.
Can you contact me by 1@sh0rt.in ?
I will then get a look over your situation and we will solve it directly. I think it´s only a little stop lost or mixed with my instruction.

lukasig, 2020-12-30:
> dd if=/mmcblk0p1 of=/dev/sda1 bs=1M
gives
> dd: can't open '/mmcblk0p1': No such file or directory

you might want to add the **dev** part in your initial setup instructions (`dd if=/dev/mmcblk0p1 of=/dev/sda1 bs=1M`)

PsyEng, 2020-12-30:
> 
> you might want to add the **dev** part in your initial setup instructions (`dd if=/dev/mmcblk0p1 of=/dev/sda1 bs=1M`)

Sure, that´s right. 
Did it work now?

lukasig, 2020-12-30:
> mount /dev/sda1 /data/media/sda1

gives

> mount: mounting /dev/sda1 on /data/media/sda1 failed: Device or resource busy

I've tried rebooting but no change...

lukasig, 2020-12-30:
Step no 7. a little confusing. After running:
`sed 's/mmcblk0p/sda/ fstab.disk"`
i get a empty prompt ">" and system does nothing....

PsyEng, 2020-12-30:
> 
> 
> Step no 7. a little confusing. After running:
> `sed 's/mmcblk0p/sda/ fstab.disk"`
> i get a empty prompt ">" and system does nothing....

Yepp, there´s a typo. I´m sorry, but it´s good someone trace back my steps, so I can see, were I missed something or anything else. 
I had to experiment a lot with all the commands, that´s why some commands are not 100% correct.
I´ve updated the command and it should look like this "sed -i 's/mmcblk0p/sda/' somefile".
> 
> also the link in step 9. is broken. I guess you were trying to link [this page](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md)?
> but then what did you mean by
> 

They´ve changed "raspberryPi" to "raspberrypi" and have case-sensitive links...
> > Create a bootable USB drive and copy all the needed files to "/data/media/sda1"
> 
> ? Should I guess you were trying to indicate how to make rpi4 boot from USB?

Yepp, thats correct. Is that so misleading? If yes, i will change it.

lukasig, 2020-12-30:
Thanks

can you provide a more detailed explanation of how to achive point 8?

lukasig, 2020-12-31:
✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅

Hey everyone

I figured it out, and it could not be any easier:

1. Flash the motioneyeOS image to your USB drive using etcher.
2. Reconnect your USB drive to your PC, go into the "boot" partition and open the file called "**cmdline.txt**".
3. Look for "**root=/dev/mmcblk0p2**" and replace it with "**root=/dev/sda2**"
4. Eject the USB drive, plug it into your rpi and enjoy :)

Notes:
- you need to make sure that your pi is able to boot from USB. If not, take a look [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md) and [here](https://www.tomshardware.com/how-to/boot-raspberry-pi-4-usb).
- make sure the sata adapter you use is compatible. Look [here](https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/). 

✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅

starbasessd, 2020-12-31:
Which version did you try with?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PsyEng, 2020-12-31:
> ✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅✅
>
> Hey everyone
>
> I figured it out, and it could not be any easier:
>
>     1. Flash the motioneyeOS image to your USB drive using etcher.
>
>     2. Reconnect your USB drive [...]

 It will work, that's correct, but it can corrupt the eeprom and mess up your booting ability. I wasn't able to boot and reboot by USB after this, only if I replugged my SSD. It doesn't matter which USB case I used, it doesn't matter which OS I used and even SD-card boot wasn't working in 3 of 4 times. 
I had to recover my eeprom, which wasn't quiet funny. 
Please investigate some more experience, before follow this guide.
I do know this is working, like I mentioned here https://github.com/ccrisan/motioneyeos/issues/2439#issuecomment-751898411

lukasig, 2020-12-31:
I use motioneyeOS version **[dev20201026](dev20201026)**
Raspberry Pi 4 4gb model.

@PsyEng 
I think your pi might have a problem
I have rebooted my pi a couple of times, and also switched from my motioneye SSD to my nextcloupi SSD a couple of times and they both boot without any problems every time.

PsyEng, 2020-12-31:
I don't think so, because I booted my Pi more then 30 times the days before and it worked fine.
There wasn't any problem in this way. The USB case didn't get power anymore, only if I replugged it.
After recoveringy my eeprom, it worked fine.

starbasessd, 2020-12-31:
@lukasig Which PI and which version of motionEyeOS are you using? I can test your process very quickly here. It's obvious to me that @PsyEng is using a Pi4, which is really a different animal than the Pi3B+ and earlier.

PsyEng, 2020-12-31:
> @lukasig Which PI and which version of motionEyeOS are you using? I can test your process very quickly here. It's obvious to me that 

It does work, I did it before and said it also long ago it does work. 
> @PsyEng is using a Pi4, which is really a different animal than the Pi3B+ and earlier.

I'm using Pi4, too. Never mentioned 3B+ or a thing else...

starbasessd, 2020-12-31:
Wait, what? @PsyEng you are replying as if you were @lukasig.

I just tried with the Pi3B+, fails to even give a rainbow screen using
dev20201026 for Pi3.
I just tried with Pi4B-4GB with dev20201026, got the first boot up
2nd boot up failed at loading dhcp for wired network.
Popped up a ping failure, then popped up a panic.
Rebooted on it's own.
Reboot loop with the above hang, ping, and panic.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

PsyEng, 2020-12-31:
@starbasessd Oh sry, I misunderstood your sentence with the Pi4. @lukasig is using the Pi4 also, I got in contact with him at Telegram.
I'm not sure, but I think the Pi3B+ is working too. Will test it next week or so, but not today, my wife is killing me 😬

Happy New Year all together 😊

lukasig, 2021-01-01:
@starbasessd 
I did answer but you did not see it...

PsyEng, 2021-01-13:
I got my new Pi4 and camera yesterday, after 14 days...
I will try to test this weekend some scenarios and hopefully not short again my camera on the aluminiumcase of the Pi, but this happened after our last conversation, so I don´t know, if this is the same problem.
It resulted in a strange behavior on the Pi, when he´s try to record, one core get in panic, uptime shows 500% load, but there´s absolutly no load on any of the cores.
So, I´ve now a completly new setup and test it out.

@lukasig For your problem with USB 3.0:
If "lsusb" is working on mOS, write down the XXXX:XXXX ID of your USB-adapter and write this into your /boot/cmdline.txt "usb-storage.quirks=XXXX:XXXX:u".
You´ve to enable the debug mode under expert settings to modify the cmdline.txt file.

starbasessd, 2021-03-26:
> 4.) Now run "dd if=/dev/mmcblk0p1 of=/dev/sda1 bs=1M", "dd
> if=/dev/mmcblk0p2 of=/dev/sda2 bs=1M", "dd if=/dev/mmcblk0p3 of=/dev/sda3
> 7.) Now navigate to "/data/media/sda2/etc" and run this commands "sed -i
> 's/mmcblk0p/sda/' fstab.disk", "sed -i 's/mmcblk0p/sda/' mtab". Now
> navigate to "/data/media/sda1" and run this command "sed -i
> 8.) Also add this line "start_x=1" to your config.txt, otherwise your MMAL
> camera won't work at all.
> 9.) Now follow this guide Create a bootable USB drive
> 10.) Now you can shutdown your Pi, eject the SD-Card and boot from your
> Edit: I added step 9, because I run a similar setup on my Pi 3, which
> didn't need the modification of the firmware files in the boot partition.
> For me this fails at step#6 with an error message that devices are busy.
> umount does not resolve the problem

Unfortunately, I've never been able to get it to run on my 4GB or 8GB Pi4s,
with 128GB USB3 thumb drive.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

lukasig, 2021-03-26:
@starbasessd

i think you're going the long and unnecessary way arround

did you take a look at [this](https://github.com/ccrisan/motioneyeos/issues/2439#issuecomment-752981870)?

starbasessd, 2021-03-26:
BTW, dev20201026 IS working on my 4GB with the cmdline.txt mod only.
The 'long way around' never did work...

lukasig, 2021-03-26:
never did work for me either. I've been using dev20201026 since it came out with no issues

ur7x, 2021-03-26:
> 
> 
> never did work for me either. I've been using dev20201026 since it came out with no issues

Yup spend a day fighting with "the long way"  
The "short way" worked on the first try.  Thanks!