motioneye-project/motioneyeos #2441: ZWO ASI 120MC-s Camera

maserowik, 2020-06-23:
I am unable to get my ASI camera talking to the motion eyes.  is this camera support in this software ?

starbasessd, 2020-06-24:
You don't say what the rest of your setup is.
If a SBC then try connecting with Raspbian to confirm the kernel supports
it.
If a PC then try using something like Ubuntu & cheese with v4l2.
It seems like it might be a USB but the descriptions didn't describe
anything like streaming like a web cam.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd