motioneye-project/motioneyeos #2442: Motioneyeos motion trigger not working

soupedsandwich, 2020-06-24:
Hi so recently I set up motioneyeos onto a Raspberry pi4, managed to get everything set up fine, I can record continuously or take still images manually and i can upload them to my google drive.

My only problem is the motion trigger option does not record any motion, I've tried making it as sensitive as I could but still no luck.

I have never used any raspberry pi boards before and dont know pretty much anything about code is there any way I can activate the motion trigger reliably?

Im using:
Motioneyeos ver 20200606
Motion ver 4.2.2+git UNKNOWN 
Motioneye ver 0.42.1
Raspberry Pi4
Raspberry Pi compatible camera MMAL

I used a ethernet cable for the first boot but I was able to get it running over wifi.

starbasessd, 2020-06-24:
If you can see the video and take snaps, should be able to record motion detection.
I put a screenshot of my settings for one of my cameras.
I also use (With Auto Threshhold Tuning Off) 1.5% or sometimes lower for Frame Change Threshhold (with 0 for  Maximum Frame Change)
I use Show Frame Changes for diagnostics to see if the camera sees any changes.
I turn on Despeckle Filter if the camera picture seems 'noisy' like static-y...
[image]

niklaslepo, 2020-07-21:
I have a problem with the camera recording everything, even though theres obiously no motion detected. It fills up my drive with 24-7 video without any motion. Anyone have good motion detection settings that really works like a real motion detecting camera?

starbasessd, 2020-07-21:
@niklaslepo:
What version of motionEyeOS?
Which Pi?
What camera(s)?
In Motion Detection, did you set
Auto Threshold 'On'?
Auto Noise Detection & Despecle Filter 'On'?
Show Frame Changes 'On'?
Most of my camera settings are
Motion Gap 2 Seconds
Captured Before 10 Frames
Captured After 10 Frames
Minimum Motion 2 Frames
I only use Mask when looking towards trees to block the movement

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

niklaslepo, 2020-07-28:
@starbasessd 
What version of motionEyeOS? 20200606
Which Pi? Rasberry Pi 3b
What camera(s)? Raspberry Pi Camera Noir
In Motion Detection, did you set
Auto Threshold 'On'? Off
Auto Noise Detection & Despecle Filter 'On'? Off
Show Frame Changes 'On'? Off

I used these settings since I saw they worked for someone else...

http://fs5.directupload.net/images/171023/k72g72bd.png

starbasessd, 2020-07-28:
The features I suggested be 'On' help eliminate 'false positives', and
using Show Frame Changes should be 'On" while setting up and testing motion
detection.
You can turn it off after you get the rest of your settings working the way
you want, to reduce the CPU/GPU overhead.
You can also try downgrading to 20190911 or 20200203, to see if that makes
any difference in your setup.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd