motioneye-project/motioneyeos #2443: All my settings disappear after i overclock my Raspberry Pi Zero W

TheLedwog, 2020-06-25:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20190911.

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: MMAL camera

My camera model is: ZeroCam - Camera for Raspberry Pi Zero

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

 

When I Overclock my Raspberry Pi Zero W I lose all my expert settings and motion detection settings. When i switch back to no overclocking i still don`t get any of my settings back and i then have to re-write the motioneyeos software back onto my raspberry pi.

MarcoMagana, 2020-06-25:
Same has happend to me a few times. I got them back by deleting the camera and adding it again.

TheLedwog, 2020-06-26:
OK thank you this worked for me.

closed 2020-06-26 by TheLedwog