motioneye-project/motioneyeos #2444: Super unreliable camera dropping out every 20 minutes

Planeguy102, 2020-06-26:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Raspberry Pi 4B

### Camera

I am using the following type of camera: MMAL

My camera model is: 5MP OV5647 Sensor

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4839232/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4839233/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4839234/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/4839235/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4839236/motioneye.log)

I'm having a very annoying issue with my camera. When I create a new install of Motioneye OS (using a Raspi 4, plugged into the official power supply, running 20190911), all seems fine. But as I start modifying settings, one section after another, at some point the camera will stop working, forcing me to reboot the system even if I changed a very minor setting (Most recent example is when I changed the size of the text.) After I had everything configured, though, with no dropouts, the camera would just turn off after about 20-30 minutes, again forcing a reboot. To top it off, it always sets the time 5 hours ahead of what it's supposed to, no matter if I use HTTP or NTP or different servers for both. I'm honestly at a loss for what could be causing it.

starbasessd, 2020-06-26:
Your dhcp server is handing out an invalid dns server. 75.75.75.75 doesn't
respond to dns lookup queries.
On your router or whatever is handing out the DHCP, try changing DNS
Servers to 4.2.4 or 8.8.8.8 or your ISP's DNS server.
Is your time zone set correctly in Expert Settings?
Do you have your PI in a case? Pi4Bs, even with the updated firmware are
notorious for overheating.
I'd recommend using heat sinks and a fan.
There are commands to run to check for under-voltage and over-heating, if
yo ulike I can post them.
Best servers for time are ntp   pool.ntp.org
More after you answer these questions.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Planeguy102, 2020-06-26:
I don't have access to my router, and it doesn't seem to be an issue right now.
There is no option for the time zone in my expert settings for some reason.
My Pi doesn't seem to be overheating but I can check if you want me to, it has heatsinks on it.
I switched it to pool.ntp.org and it's now 5 hours ahead.
I'm considering updating the version to 202006.

starbasessd, 2020-06-26:
20200206 is now considered latest, but has several issues, one of which is
'continuous recording'.
I am installing on my Pi4B-8GB to check for time zone settings.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Planeguy102, 2020-06-26:
I only need motion detection recording, that's no issue. But I've been experiencing this issue with both my Pi4 and Pi3B, and I'm fed up with it.

starbasessd, 2020-06-26:
My bad. Time Zone is under General
Under Expert
Date Method  SNTP
NTP Server pool.ntp.org

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-06-26:
To check if the PI is reporting low voltage:
SSH or console (not GUI)

vcgencmd get_throttled

If it reports 0x0 you are good to go
To check overheat:
vcgencmd measure_temp
This will show CPU/GPU temp. Anything over 50 is bad. Over 60 and the
processor will pause (throttle) and cause issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Planeguy102, 2020-06-26:
It tells me it's not throttled and it's sitting at a chilly 39C. I left it running with my other fisheye sensor and it still dropped out in less than an hour.

starbasessd, 2020-06-26:
Can you go back to a clean local only install, without trying to save to
the SMB server?
Your motioneye log shows failure to save after a while.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Planeguy102, 2020-06-26:
I am currently testing a clean 202006 install and it has yet to die, so I have to assume it doesn't really like my windows share drive.

starbasessd, 2020-06-26:
Or something on your network is causing it to drop the connection...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Planeguy102, 2020-06-26:
The network hypothesis is unlikely, as I can still change settings and SSH into the Pi after the camera drops out.

Planeguy102, 2020-06-26:
No, yeah, I set it to the fileshare, and it cut out after a bit, and as soon as I switched it to the "Custom Path" option, it turned right back on. That's rather unfortunate.

closed 2020-06-26 by Planeguy102