motioneye-project/motioneyeos #2445: No General Settings

patnan, 2020-06-26:
When I start a fresh motioneye I get no General Settings when I log in.

----
I am running motionEyeOS version: various from 2018 including latest 20200606

I am using the following board/model: tried with Pi1 as well as Pi3

I am using the Pi NOIR module v2 

My motionEyeOS unit is connected to the network via:  Ethernet

[image]

starbasessd, 2020-06-26:
Are you logged in as User, or Admin?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

patnan, 2020-06-27:
Regardless if I log in as Admin (blank pw), user (blank pw) or just click login i get the same behavior.

When logging in as admin, I have to do it twice since i get two prompts.

starbasessd, 2020-06-27:
You are using
https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi-20200606.img.xz
for your Pi1, and
https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-raspberrypi3-20200606.img.xz
for your Pi3?
I have tested both on a clean MicroSD card (for the PI1 including the
proper adapter) and both perform normally.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

patnan, 2020-06-27:
Yes

I know it's wierd, that's why I ask here.

patnan, 2020-06-27:
Had a trailing whitespace after "admin". How it works.

closed 2020-06-27 by patnan