motioneye-project/motioneyeos #2446: Can't access Raspberry Pi 0 W after adding wifi settings through motioneyeos

maximumentropy1, 2020-06-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi-20190911.img

### Board Model

I am using the following board/model: Raspberry pi 0 W

### Camera

I am using the following type of camera: Simple MJPEG (I think - whatever the pi camera is)

My camera model is: Raspberry Pi NoIR Camera Board v2

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet, but when I add the wifi configuration it stops working

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * Micro USB to USB to Ethernet adapter

### Log Files

I consider the following log files relevant to this issue:

 * I cannot access the log files

starbasessd, 2020-06-28:
PiCam is type MMAL.
Can you hook a monitor to it to watch the reboot following the wifi changes?
Can you just add the wpa_supplicant.conf file to the /boot partition right
after imaging, but before you insert into the PI before first boot?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

maximumentropy1, 2020-06-28:
So I already tried adding wpa_supplicant.conf files twice, both times it did connect to the wifi, but I was unable to access it. I also tried using ssh, but it just says no route to host.

starbasessd, 2020-06-28:
You need to see what the IP address is for the wifi.
You can ssh into the ethernet IP Address, then run command:
ifconfig <enter>
It will list the IP Address for the ethernet and wifi adapters.
You can force a name for the device by adding the file hostname to /boot
with something easy to remember, like motioneye
After booting, try ssh to motioneye.local.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

maximumentropy1, 2020-06-28:
Now it just says connection refused. I thought MotionEyeOS automatically enables ssh?

starbasessd, 2020-06-28:
No. Need to add a file in /boot called ssh or ssh.txt. Doesn't need
anything in it.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-06-28:
Were you able to get into the Pi as expected?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

maximumentropy1, 2020-06-28:
No, I was never able to figure out why I couldn't access it after adding the wireless configuration. It's like that broke it somehow. I think I am going to try and use raspbian instead, because that seems more stable.

closed 2020-06-28 by maximumentropy1