motioneye-project/motioneyeos #2447: Add audio support on video and sound trigger for video record

userosos, 2020-06-29:
Hello! What about audio record? I use ONVIF camera it has microphone but motioneyeos not record audio with video. I think that if will be add triggired to record video from audio noizes in an microphone alos. It will be good.
Thx for atthenhion!

Fallen63, 2020-07-21:
It's a very good idea.
I am also interested.
thank you for your work

originalknibbler, 2020-10-24:
Hi audio support would be perfect then it would possible to use it as a babyphone. Or maybe to start the recording with noise and/or motion :-) But thank you itś a nice OS so far.

jasaw, 2020-10-25:
Audio is the most requested feature, but we rely on Motion software to do motion processing and video recording, and Motion developers are reluctant to support audio recording. Even if we implement it ourselves, it will be too much work maintaining it if it's not merged upstream (into Motion project).
See this original issue for more info:
https://github.com/ccrisan/motioneyeos/issues/105