motioneye-project/motioneyeos #2448: High Quality Pi Camera - Unable to open video device

BruffDoraleous, 2020-07-01:
Hi, I'm pretty new to motioneyos.  I'm using the raspberry pi 4B-8gb model.  The pi is connected via Ethernet.  I'm using the new raspberry high quality camera (and think this may be part of the issue).  When selecting local V4L2 camera the widow is greyed out and says unable to open video device.  I tried using MMAL but there were no camera options to select.  Is this a compatibility issue with the new high quality camera?  I have reflashed the ssd with the newest version multiple times, updated the firmware, verified that the high quality camera works. Thanks in advance for any input.

starbasessd, 2020-07-01:
You don't say which version of motioneyeos.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BruffDoraleous, 2020-07-01:
Sorry I'm using version 20190911.  I needed to use this fix https://github.com/ccrisan/motioneyeos/issues/2419 in order to get it to boot.  After copying fixup4.dat and start4.elf from previous versions the pi booted but will not connect/detect the camera.

starbasessd, 2020-07-01:
The new HD camera wasn't supported in the kernel back then, it didn't exist
yet. You could try 20200203 or 20200606 with that same boot fix, but no
guarantee, as the driver may not be in the kernel @CCRISAN used.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BruffDoraleous, 2020-07-01:
I tried both versions and still have the same issue.  I'll assume I just needs to wait until @ccrisan adds the HD driver.  Thanks!

starbasessd, 2020-07-01:
You can verify by installing Raspbian/RaspberryPiOS dated after the release
of the HD Camera and confirming it works there (The Latest would be fine
*2020-05-27* and do a full update)
Just to make sure there isn't a problem with the hardware...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

BruffDoraleous, 2020-07-01:
I have already done this and have taken some pretty great stills so it's not a hardware issue.  Thanks for the input.

starbasessd, 2020-07-01:
Then it is just a matter of time for @CCRISAN to generate the next version.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

koenvp, 2020-08-18:
Same issue here.
Raspberry pi 4B-8GB + the official "High Quality Pi Camera".
I'm on motionEyeOs 20200606

When running Raspberry Pi OS, I have no problem making stills and video's with the HQ camera.
When running motionEyeOs, I'm not able to get any image/stream/video/still from the camera.

Is there any update on this issue or a known workaround for now?

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd