motioneye-project/motioneyeos #2449: date format DD/MM/YY but movie files not in right order

SivanDuijn, 2020-07-01:
I've set my movies storage file names to be [%d-%m-%Y/%H-%M-%S], that works wonderfully.
But when opening the recorded movies window, it doesn't open the folder for this day by default
[image]

I'd like it to open the 01-07-2020 folder.
Ofcourse this is happening because 30 is bigger than 01 but could there be a solution for this?

motionEye Version: 0.42.1
Motion Version:      4.2.2+gitUNKNOWN
OS Version:             motionEyeOS 20200606

starbasessd, 2020-07-01:
I use: [%Y/%m%d/%H-%M-%S] sorts it into folder: Year, folder month, folder
day H:M:S.

starbasessd, 2020-10-23:
Closing, no comments/requests/followup after 60 days
If the issue is still not resolved, feel free to re-open or open a new case.

closed 2020-10-23 by starbasessd